RTimothyEdwards/netgen #2: [OpenBSD/ld.bfd] (netgen|magic)exec wants GR_LIBS to link successfully

julianaito, 2020-02-17:
Hi!

While netgen and magic builds properly with LLVM's lld on OpenBSD, our older GNU binutils ld.bfd [fails while linking netgenexec](http://build-failures.rhaalovely.net/powerpc/last/cad/netgen.log).

Note that magic is also impacted because they share the same build script.

Adding GR_LIBS to netgenexec lib flags does the job. I'm not sure it's the right thing to do, so i wanted to know you opinion about that.

(cc @just22 @kmosiejczuk)

```diff
--- tcltk/Makefile.orig
+++ tcltk/Makefile
@@ -25,7 +25,7 @@ install-tcl: netgenexec${EXEEXT} $(DESTDIR)${BINDIR}/n
 netgenexec${EXEEXT}: netgenexec.c
 	${CC} ${CFLAGS} ${CPPFLAGS} ${DFLAGS_NOSTUB} netgenexec.c \
 		-o netgenexec${EXEEXT} \
-		${LDFLAGS} ${LIBS} ${LIB_SPECS_NOSTUB}
+		${LIB_SPECS_NOSTUB} ${LDFLAGS} ${LIBS} ${GR_LIBS}
 
 netgen.tcl: netgen.tcl.in
 	sed -e 's%TCL_DIR%${TCLDIR}%g' \
```

RTimothyEdwards, 2020-02-17:
For me, doing "ldd /usr/local/lib/netgen/tcl/netgenexec" shows that libX11 is linked to the file, so I assume that there is no harm in doing this.

RTimothyEdwards, 2020-02-17:
Pushed fix to opencircuitdesign.com ;  this will be updated in github overnight.  I will close this tomorrow after the update.

julianaito, 2020-02-17:
Thanks a lot! I'll provide a similar change for magic soon.

closed 2020-02-18 by RTimothyEdwards