RTimothyEdwards/netgen #3: [1.6.3] warning: multiple unsequenced modifications to 'ActelIndex'

yurivict, 2020-03-21:
clang-8 complains:
```
actel.c:93:19: warning: multiple unsequenced modifications to 'ActelIndex' [-Wunsequenced]
    ActelIndex = (++ActelIndex) % ACTELNAMESIZE;
               ~  ^
actel.c:104:17: warning: multiple unsequenced modifications to 'ActelIndex' [-Wunsequenced]
  ActelIndex = (++ActelIndex) % ACTELNAMESIZE;
             ~  ^
```

The order of modifications is undefined and these lines are invalid.

RTimothyEdwards, 2020-03-22:
There was one other warning issue you submitted (in qflow, I think) that was a real bug and needed to be fixed.  This one, however, I must assert that I find absolutely nothing wrong with the statement. It is syntactically unambiguous.  I would submit this as a  bug report to the clang developers.  This sort of warning should be issued for ambiguous (or just incorrect) statements like "a = a++".  In the clang test at https://github.com/llvm-mirror/clang/blob/master/test/SemaCXX/warn-unsequenced.cpp, the closest similar thing is "a = f(++a)" and is said to be okay.

dcb314, 2020-09-15:
>This one, however, I must assert that I find absolutely nothing wrong with the statement.

Then you need to understand C better.
Incrementing and writing to a variable between sequence points is a common error.

The clang developers have a very good understanding of C and I'd be extremely careful indeed
in calling into question their C knowledge.

Interestingly, gcc also finds the problem for this code:

#define ACTELNAMESIZE 10

extern void g( int ActelIndex);

void f( int ActelIndex)
{
	ActelIndex = (++ActelIndex) % ACTELNAMESIZE;
	g( ActelIndex);
}

$ gcc -c -Wall -O2 /tmp/q.c
/tmp/q.c: In function ‘f’:
/tmp/q.c:8:20: warning: operation on ‘ActelIndex’ may be undefined [-Wsequence-point]
    8 |         ActelIndex = (++ActelIndex) % ACTELNAMESIZE;
      |                    ^