RTimothyEdwards/netgen #4: install fails: cp: netgen: No such file or directory

yurivict, 2020-03-21:
```
--- installing executable to /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin
--- installing run-time files to /usr/ports/cad/netgen-lvs/work/stage/usr/local/lib
cp: netgen: No such file or directory
gmake[3]: *** [Makefile:30: /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin/netgen] Error 1
cp: netgen: No such file or directory
gmake[3]: *** [Makefile:30: /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin/netgen] Error 1
```

Version 1.6.3
FreeBSD 12

RTimothyEdwards, 2020-03-22:
What did your "configure" statement look like?  Were there any error messages during the compile?

yurivict, 2020-03-22:
```
$ ./configure --prefix=/usr/local --localstatedir=/var --mandir=/usr/local/man --infodir=/usr/local/share/info/ --build=amd64-portbld-freebsd12.0
```

```
===>  Configuring for netgen-lvs-1.6.3
configure: loading site script /usr/ports/Templates/config.site
checking build system type... amd64-portbld-freebsd12.0
checking host system type... amd64-portbld-freebsd12.0
checking target system type... amd64-portbld-freebsd12.0
checking for gcc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether cc accepts -g... yes
checking for cc option to accept ISO C89... none needed
checking how to run the C preprocessor... cpp
checking for library containing strerror... none required
checking for a BSD-compatible install... /usr/bin/install -c
checking for ranlib... ranlib
checking for gm4... /usr/local/bin/gm4
checking for ld used by GCC... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for egrep... (cached) /usr/bin/egrep
checking for ANSI C header files... (cached) yes
checking for sys/types.h... (cached) yes
checking for sys/stat.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for memory.h... (cached) yes
checking for strings.h... (cached) yes
checking for inttypes.h... (cached) yes
checking for stdint.h... (cached) yes
checking for unistd.h... (cached) yes
checking size of void *... 8
checking size of unsigned int... 4
checking size of unsigned long... 8
checking size of unsigned long long... 8
checking whether byte ordering is bigendian... no
checking for ANSI C header files... (cached) yes
checking for setenv... (cached) yes
checking for putenv... (cached) yes
checking for vfork... (cached) yes
checking for dirent.h... (cached) yes
checking for limits.h... (cached) yes
checking param.h usability... no
checking param.h presence... no
checking for param.h... no
checking for va_copy... yes
checking for __va_copy... yes
checking for tclConfig.sh... 
Can't find Tcl configuration script "tclConfig.sh"
Reverting to non-Tcl compilation
checking for X... libraries /usr/X11R6/lib, headers /usr/X11R6/include
checking whether -R must be followed by a space... neither works
checking for gethostbyname... (cached) yes
checking for connect... yes
checking for remove... yes
checking for shmat... yes
checking for IceConnectionNumber in -lICE... yes
configure: creating ./config.status
config.status: creating defs.mak
config.status: WARNING:  'defs.mak.in' seems to ignore the --datarootdir setting

-----------------------------------------------------------
Configuration Summary (principle requirements):
[... 101 lines cut ...]
  if ((ptr->class == CLASS_SUBCKT)) PrintLeavesInCell(ptr->name, ptr->file);
      ~           ^              ~
query.c:926:19: note: use '=' to turn this equality comparison into an assignment
  if ((ptr->class == CLASS_SUBCKT)) PrintLeavesInCell(ptr->name, ptr->file);
                  ^~
                  =
2 warnings generated.
gmake[3]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3/base'
gmake[2]: *** [Makefile:44: modules] Error 2
gmake[2]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
gmake[1]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
===>  Staging for netgen-lvs-1.6.3
===>   netgen-lvs-1.6.3 depends on file: /usr/local/bin/python3.7 - found
===>   Generating temporary packing list
gmake[1]: Entering directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
--- installing executable to /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin
--- installing run-time files to /usr/ports/cad/netgen-lvs/work/stage/usr/local/lib
cp: netgen: No such file or directory
gmake[3]: *** [Makefile:30: /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin/netgen] Error 1
cp: netgen: No such file or directory
gmake[3]: *** [Makefile:30: /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin/netgen] Error 1
gmake[1]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
====> Compressing man pages (compress-man)
====> Running Q/A tests (stage-qa)
```

yurivict, 2020-03-22:
When I added Tcl/Tk dependencies, it has another problem:
```
===>  Configuring for netgen-lvs-1.6.3
configure: loading site script /usr/ports/Templates/config.site
checking build system type... amd64-portbld-freebsd12.0
checking host system type... amd64-portbld-freebsd12.0
checking target system type... amd64-portbld-freebsd12.0
checking for gcc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether cc accepts -g... yes
checking for cc option to accept ISO C89... none needed
checking how to run the C preprocessor... cpp
checking for library containing strerror... none required
checking for a BSD-compatible install... /usr/bin/install -c
checking for ranlib... ranlib
checking for gm4... /usr/local/bin/gm4
checking for ld used by GCC... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for egrep... (cached) /usr/bin/egrep
checking for ANSI C header files... (cached) yes
checking for sys/types.h... (cached) yes
checking for sys/stat.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for memory.h... (cached) yes
checking for strings.h... (cached) yes
checking for inttypes.h... (cached) yes
checking for stdint.h... (cached) yes
checking for unistd.h... (cached) yes
checking size of void *... 8
checking size of unsigned int... 4
checking size of unsigned long... 8
checking size of unsigned long long... 8
checking whether byte ordering is bigendian... no
checking for ANSI C header files... (cached) yes
checking for setenv... (cached) yes
checking for putenv... (cached) yes
checking for vfork... (cached) yes
checking for dirent.h... (cached) yes
checking for limits.h... (cached) yes
checking param.h usability... no
checking param.h presence... no
checking for param.h... no
checking for va_copy... yes
checking for __va_copy... yes
checking for tclConfig.sh... /usr/local/lib/tcl8.6/tclConfig.sh
checking for tkConfig.sh... /usr/local/lib/tk8.6/tkConfig.sh
checking for wish executable... /usr/local/bin/wish8.6
checking for tclsh executable... /usr/local/bin/tclsh8.6
checking for X... libraries /usr/X11R6/lib, headers /usr/X11R6/include
checking whether -R must be followed by a space... neither works
checking for gethostbyname... (cached) yes
checking for connect... yes
checking for remove... yes
checking for shmat... yes
checking for IceConnectionNumber in -lICE... yes
configure: creating ./config.status
config.status: creating defs.mak
config.status: WARNING:  'defs.mak.in' seems to ignore the --datarootdir setting

-----------------------------------------------------------
Configuration Summary (principle requirements):

Tcl/Tk:      yes
[... 170 lines cut ...]
	-o netgenexec \
	 -fstack-protector-strong   -lm -L/usr/local/lib -L/usr/local/lib -ltk86 -L/usr/local/lib -ltcl86
sed -e /TCL_DIR/s%TCL_DIR%/usr/local/lib/netgen/tcl%g \
    -e /SHDLIB_EXT/s%SHDLIB_EXT%.so%g \
    netgen.tcl.in > netgen.tcl
sed -e /TCL_DIR/s%TCL_DIR%/usr/local/lib/netgen/tcl%g \
    -e /TCLLIB_DIR/s%TCLLIB_DIR%/usr/local/lib%g \
    -e /WISH_EXE/s%WISH_EXE%/usr/local/bin/wish8.6%g \
    netgen.sh.in > netgen.sh
gmake[3]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3/tcltk'
gmake[2]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
gmake[1]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
===>  Staging for netgen-lvs-1.6.3
===>   netgen-lvs-1.6.3 depends on file: /usr/local/bin/python3.7 - found
===>   Generating temporary packing list
gmake[1]: Entering directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
./scripts/mkdirs /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin /usr/ports/cad/netgen-lvs/work/stage/usr/local/man
--- installing executable to /usr/ports/cad/netgen-lvs/work/stage/usr/local/bin
--- installing run-time files to /usr/ports/cad/netgen-lvs/work/stage/usr/local/lib
gmake[3]: *** No rule to make target '../base/libbase.o', needed by 'tclnetgen.so'.  Stop.
gmake[1]: Leaving directory '/usr/ports/cad/netgen-lvs/work/netgen-1.6.3'
====> Compressing man pages (compress-man)
====> Running Q/A tests (stage-qa)
Warning: 'lib/netgen/tcl/netgenexec' is not stripped consider trying INSTALL_TARGET=install-strip or using ${STRIP_CMD}
```

RTimothyEdwards, 2020-03-22:
This compile makes no sense in the context of the github repo.  It is for "netgen-lvs-1.6.3".  I recall that the packaged version was renamed from "netgen" to "netgen-lvs" because there was already some package named "netgen" for 3D mesh generation (which annoys me because this version predates it by decades).  But I don't know who decided on the "1.6.3" designation.  I'm on 1.5.144.  And the point where the compile is failing, "hash.c:317:30: error: non-void function 'HashKill' should return a value [-Wreturn-type]", in my code HashKill is a void function, so this is not an error.  If you wish to compile netgen, I suggest using the source code in this repository.

RTimothyEdwards, 2020-03-22:
The errors in your previous posting are of a similar nature;  the error message complains about no return value in a function returning int, but my code is definitely returning a value in those places.

yurivict, 2020-03-22:
There might be some confusion. The tarballs are taken from http://opencircuitdesign.com/netgen/archive/ Is this not the right place? Are these different projects?

yurivict, 2020-03-22:
I switched the FreeBSD to GitHub's version ```1.5.144``` and the problems disappeared.
It seems that http://opencircuitdesign.com/netgen/archive contains some stray, outdated tarballs with higher version numbers.

closed 2020-03-22 by yurivict

RTimothyEdwards, 2020-03-22:
You're right;  my apologies.  I don't even remember starting a 1.6 development branch!  The files are dated April and August 2017, so yes, they are pretty far out of date.  Sorry about the confusion.  I even confused myself!

yurivict, 2020-03-22:
No problem!