RTimothyEdwards/netgen #8: netgen doesn't appear to have standard --help or --version?

mithro, 2020-11-28:
Installed netgen via `apt-get install netgen-lvs` and ended up with the following version `Netgen 1.5.133 compiled on Sun Feb  2 17:09:39 UTC 2020`

This version doesn't seem to have any `--help` or `--version`?

RTimothyEdwards, 2020-11-28:
So what am I supposed to do about it?  I don't know how netgen gets into a packaged version.

mithro, 2020-11-28:
I just installed the latest version of netgen and it still doesn't seem to have any `--help` or `--version` command line arguments?

RTimothyEdwards, 2020-11-28:
@mithro :  Did somebody compile it without Tcl/Tk support?

mithro, 2020-11-28:
Seems to have tcl/tk8 enabled?

mithro, 2020-11-28:
When I do a `/opt/magic/bin/netgen --version` it pops up a GUI with a tcl/tk console thingy...

RTimothyEdwards, 2020-11-28:
@mithro :  Well, actually, I guess it doesn't support --version or --help.  All of the programs I develop are old, and it's hard to keep track of what things I've done to support standard behavior. . .

RTimothyEdwards, 2020-11-28:
@mithro :  So leave this issue up and I'll get around to it sooner or later.