RTimothyEdwards/netgen #9: Why is x11 required for tcl support?

mithro, 2020-11-30:
The tcl console seems to run fine without an X server. Is there any reason netgen seems to require X11 support to enable it?

RTimothyEdwards, 2020-11-30:
The Tcl console cannot run without an X server.  Where did you get the idea that it does?

mithro, 2020-11-30:
The fact that I don't need X to run tcl scripts in Magic?

RTimothyEdwards, 2020-11-30:
That has nothing to do with the console.  Both magic and netgen can be run with -noconsole (magic also needs -dnull) and you can run batch scripts.

RTimothyEdwards, 2020-11-30:
@mithro :  Maybe you're confused about what the console is.  Both magic and netgen are built as extensions to Tcl.  You can run both directly from a terminal.  The console is a Tk application where text from the terminal is redirected.

mithro, 2020-11-30:
Coming from a Python world, "the console" is when you type `python` and get the interactive `>>>` prompt...

Anyway, the real question is, why does netgen require X11 to support tcl scripting?

xobs, 2023-02-24:
I remove the X11 check in Conda on macos in order to get TCL working where X11 doesn't exist. There are no X11 headers in Conda, so this is required to get it to build.

The patch is at https://github.com/xobs/conda-eda/blob/7cc4a9fbd4aaff0e958ba63f67b0d10a82dc1f9b/misc/netgen/remove-x11-requirement.patch and seems to work well. I've managed to do a full end-to-end build on an ARM64 Mac using this patch complete with LVS.

RTimothyEdwards, 2023-02-24:
I think the short-ish answer to this is that there was an old X11 interface to netgen called `xnetgen` with C code in base/xnetgen.c, and although that is not even compiled when using the Tcl interface, all the X11 dependencies have not been pulled out of the configuration.  I think it should be feasible to do this.

RTimothyEdwards, 2023-02-24:
Okay, if after doing "configure" I remove the two library references to "tk" from definitions LIB_SPECS and LIB_SPECS_NOSTUB and references to "X11" in GR_DFLAGS and GR_LIBS in file `defs.mak`, it builds correctly without X11 or Tk.

Now I just need to pull those out of the configure script. . .

RTimothyEdwards, 2023-02-24:
@mithro :  Sorry about the delay, since I seem to have missed your last post and then forgot about the whole thing for a while. But there was a wayward setting in the configure script that incorrectly enables X11 in response to compiling with Tcl support, and a failure to check for X11 being enabled before adding Tk libraries to the linker options.  With those two simple changes, netgen (or more precisely, netgenexec) no longer links to X11 and Tk libraries.  Fixed in version 1.5.245, pushed to opencircuitdesign.com and will be mirrored to github overnight.

closed 2023-02-24 by RTimothyEdwards