RTimothyEdwards/netgen #17: Netgen isn't case sensitive

Manarabdelaty, 2021-01-28:
In the attached test case, the design has two ports with the same name but one is uppercase and the other is lower case. Netgen reports one pin mismatch because it sees both ports as lower-case.

```
output p; 
output P;
```

[spm.zip](https://github.com/RTimothyEdwards/netgen/files/5887988/spm.zip)

RTimothyEdwards, 2021-01-28:
This is a problem with the SPICE format, not with netgen.  SPICE is case insensitive.  So if you wish to compare a SPICE netlist to a verilog netlist, then the verilog netlist has to be case insensitive as well.  You can subvert this by running the LVS with the netlists swapped ("SVL", I guess. . .).  By reading the verilog first, it will force SPICE to be case sensitive.  But of course, that may have its own issues.

Manarabdelaty, 2021-01-28:
In the SPICE netlist, I can see that one port is uppercase and the other is lowercase. So, I think in this example the SPICE is case sensitive. 

```
.subckt spm P clk p rst x[0] x[10] x[11] x[12] x[13] x[14] x[15] x[16] x[17] x[18]
+ x[19] x[1] x[20] x[21] x[22] x[23] x[24] x[25] x[26] x[27] x[28] x[29] x[2] x[30]
+ x[31] x[3] x[4] x[5] x[6] x[7] x[8] x[9] y VPWR VGND
```

RTimothyEdwards, 2021-01-28:
SPICE is _by definition_ case-insensitive, so the above is an invalid SPICE netlist.