RTimothyEdwards/netgen #21: Disconnected pin not matching between two circuits

donn, 2021-07-01:
Netgen Version: 738c1f7b3705bca2f1cc66fbc1cfb20f12d49a06

So the spice model (extracted with magic) has a disconnected pin by design, but when Netgen runs it against the powered netlist (which also has a disconnected pin,) the report shows that the pin straight up `does not exist` in Circuit 1 in the log:

```
VGND                                       |VGND                                       
VPWR                                       |VPWR                                       
(no matching pin)                          |rounds[0]                                  
----------------------
```

I'm not sure if this is intended behavior

[lvs.log](https://github.com/RTimothyEdwards/netgen/files/6747734/lvs.log)
[reproducible.tar.gz](https://github.com/RTimothyEdwards/netgen/files/6747726/reproducible.tar.gz)

RTimothyEdwards, 2021-07-02:
@donn :  On an unrelated note, you have a roundabout way of launching netgen.  "-batch" is followed by a netgen command, so you are using "-batch" to run the Tcl "source" command in netgen to read a script from a file that contains a netgen command.  It would be a lot simpler to just run "netgen -batch lvs <arguments>".

RTimothyEdwards, 2021-07-02:
FYI:  *this* is what I would consider a reproducible example.  Using a 13MB example file to demonstrate a simple issue is generally frowned upon.

[simple.tar.gz](https://github.com/RTimothyEdwards/netgen/files/6754985/simple.tar.gz)

RTimothyEdwards, 2021-07-02:
So, looking into this, I see that netgen does not say the circuit has failed pin matching, so this is intended behavior and everything is working correctly.

RTimothyEdwards, 2021-07-02:
Seems like the LVS error counter script that analyzes the JSON output file would be better to move from the qflow repository to the netgen repository.  In that case, OpenLane would not need to maintain its own copy of the script.

donn, 2021-07-02:
> @donn : On an unrelated note, you have a roundabout way of launching netgen. "-batch" is followed by a netgen command, so you are using "-batch" to run the Tcl "source" command in netgen to read a script from a file that contains a netgen command. It would be a lot simpler to just run "netgen -batch lvs ".
> FYI: this is what I would consider a reproducible example. Using a 13MB example file to demonstrate a simple issue is generally frowned upon.

I put in an effort to just package OpenLane files with as little changed as possible. Apologies for the idiosyncracies.

If it's intended behavior, then that's fine. I'll update the script.

closed 2021-07-02 by donn