RTimothyEdwards/netgen #22: Parallel FETs not being merged when drain/source connections are swapped

mooredan, 2021-07-10:
Hi Tim,  I _think_ that this was working at some point in time.  Test case provided, 

[netgen_tc1.tar.gz](https://github.com/RTimothyEdwards/netgen/files/6795267/netgen_tc1.tar.gz)

this is an inverter with three parallel pfets and three parallel nfets.  

schematic:
[image]

layout:
[image]

schematic netlist:
```
.subckt inv_c z a vdd vss
M1000 z a vdd vdd pfet W=5.85u L=0.6u M=3
M1001 z a vss vss nfet w=2.75u l=0.6u m=3
.ends
```

layout netlist:
```
.subckt inv_c z a vdd vss
M1000 vss a z vss nfet w=2.75u l=0.6u
+  ad=2.29167p pd=5.33333u as=2.29167p ps=5.33333u
M1001 z a vdd vdd pfet w=5.85u l=0.6u
+  ad=4.875p pd=9.46667u as=4.875p ps=9.46667u
M1002 z a vss vss nfet w=2.75u l=0.6u
+  ad=2.29167p pd=5.33333u as=2.29167p ps=5.33333u
M1003 z a vdd vdd pfet w=5.85u l=0.6u
+  ad=4.875p pd=9.46667u as=4.875p ps=9.46667u
M1004 z a vss vss nfet w=2.75u l=0.6u
+  ad=2.29167p pd=5.33333u as=2.29167p ps=5.33333u
M1005 vdd a z vdd pfet w=5.85u l=0.6u
+  ad=4.875p pd=9.46667u as=4.875p ps=9.46667u
C0 vdd vss 4.52fF
.ends
```

netgen indicates a mismatch:

Circuit 1 contains 4 devices, Circuit 2 contains 2 devices. *** MISMATCH ***
Circuit 1 contains 4 nets,    Circuit 2 contains 4 nets.
Result: Netlists do not match.

This mismatch appears to come from the layout netlist being collapsed into 2 pfets and 2 nfets (versus one each), and  appears to be done so because the source and drain connections are swapped on one of each of the pfets and nfets.

I tested this by manually editting the layout netlist to swap these around, and LVS passed.

Am I missing a configuration setting?

RTimothyEdwards, 2021-07-10:
Your main problem is that you have allowed parasitic capacitances in your layout extraction, so you have a capacitor device in the layout that does not exist in the schematic.  If I simply comment out the "C0" line in the layout, I get a match.  But with it in, I get a different output with "Circuit 1 contains 3 devices".  Maybe you are passing it a bad setup file?  With no setup file, permutations would be allowed by default on all transistor sources and drains.

mooredan, 2021-07-10:
Thanks for the quick response.  Unfortunately, I'm unable to duplicate your results by just commenting out the C0 instance in the inv_c.cir file:

```
...
ircuit 1 contains 4 devices, Circuit 2 contains 2 devices. *** MISMATCH ***
Circuit 1 contains 4 nets,    Circuit 2 contains 4 nets.

Result: Netlists do not match.
Logging to file "lvs.log" disabled
LVS Done.
```

I also get an error:

`Error setup.tcl:1 (ignoring), No such cell!`

This is because the setup.tcl file (included in the tarball, and explicitly called out in the lvs command)  has the line to ignore the capacitors:

```
$ cat setup.tcl 
ignore class c
property nfet remove as ad ps pd
property pfet remove as ad ps pd
```
So I'll delete the "ignore class c" and rerun:

The Error message goes away, but the netlist still do not match:
```
$ ./runme.sh 
Netgen 1.5.193 compiled on Fri Jul  9 03:31:05 PM PDT 2021
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
Reading netlist file inv_c.cir
Reading netlist file inv_c.sp
Reading setup file setup.tcl
No property as found for device nfet
No property ad found for device nfet
No property ps found for device nfet
No property pd found for device nfet
No property as found for device pfet
No property ad found for device pfet
No property ps found for device pfet
No property pd found for device pfet
Comparison output logged to file lvs.log
Logging to file "lvs.log" enabled
Contents of circuit 1:  Circuit: 'inv_c'
Circuit inv_c contains 6 device instances.
  Class: pfet                  instances:   3
  Class: nfet                  instances:   3
Circuit contains 4 nets.
Contents of circuit 2:  Circuit: 'inv_c'
Circuit inv_c contains 2 device instances.
  Class: pfet                  instances:   1
  Class: nfet                  instances:   1
Circuit contains 4 nets.

Circuit was modified by parallel/series device merging.
New circuit summary:

Contents of circuit 1:  Circuit: 'inv_c'
Circuit inv_c contains 4 device instances.
  Class: pfet                  instances:   2
  Class: nfet                  instances:   2
Circuit contains 4 nets.
Contents of circuit 2:  Circuit: 'inv_c'
Circuit inv_c contains 2 device instances.
  Class: pfet                  instances:   1
  Class: nfet                  instances:   1
Circuit contains 4 nets.

Circuit 1 contains 4 devices, Circuit 2 contains 2 devices. *** MISMATCH ***
Circuit 1 contains 4 nets,    Circuit 2 contains 4 nets.

Result: Netlists do not match.
Logging to file "lvs.log" disabled
LVS Done.

```

The only way I've been able to get a is by swapping around the source and drain connections on instances M1000 and M1005 in the inv_c.cir file (and it doesn't matter if the C0 instance is there or not):

```
$ cat inv_c.cir 
* NGSPICE file created from inv_c.ext - technology: amic5n

.subckt inv_c z a vdd vss
M1000 z a vss vss nfet w=2.75u l=0.6u
+  ad=2.29167p pd=5.33333u as=2.29167p ps=5.33333u
M1001 z a vdd vdd pfet w=5.85u l=0.6u
+  ad=4.875p pd=9.46667u as=4.875p ps=9.46667u
M1002 z a vss vss nfet w=2.75u l=0.6u
+  ad=2.29167p pd=5.33333u as=2.29167p ps=5.33333u
M1003 z a vdd vdd pfet w=5.85u l=0.6u
+  ad=4.875p pd=9.46667u as=4.875p ps=9.46667u
M1004 z a vss vss nfet w=2.75u l=0.6u
+  ad=2.29167p pd=5.33333u as=2.29167p ps=5.33333u
M1005 z a vdd vdd pfet w=5.85u l=0.6u
+  ad=4.875p pd=9.46667u as=4.875p ps=9.46667u
C0 vdd vss 4.52fF
.ends
```

```
$ ./runme.sh 
Netgen 1.5.193 compiled on Fri Jul  9 03:31:05 PM PDT 2021
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
Reading netlist file inv_c.cir
Reading netlist file inv_c.sp
Reading setup file setup.tcl
No property as found for device nfet
No property ad found for device nfet
No property ps found for device nfet
No property pd found for device nfet
No property as found for device pfet
No property ad found for device pfet
No property ps found for device pfet
No property pd found for device pfet
Comparison output logged to file lvs.log
Logging to file "lvs.log" enabled
Contents of circuit 1:  Circuit: 'inv_c'
Circuit inv_c contains 6 device instances.
  Class: pfet                  instances:   3
  Class: nfet                  instances:   3
Circuit contains 4 nets.
Contents of circuit 2:  Circuit: 'inv_c'
Circuit inv_c contains 2 device instances.
  Class: pfet                  instances:   1
  Class: nfet                  instances:   1
Circuit contains 4 nets.

Circuit was modified by parallel/series device merging.
New circuit summary:

Contents of circuit 1:  Circuit: 'inv_c'
Circuit inv_c contains 2 device instances.
  Class: pfet                  instances:   1
  Class: nfet                  instances:   1
Circuit contains 4 nets.
Contents of circuit 2:  Circuit: 'inv_c'
Circuit inv_c contains 2 device instances.
  Class: pfet                  instances:   1
  Class: nfet                  instances:   1
Circuit contains 4 nets.

Circuit 1 contains 2 devices, Circuit 2 contains 2 devices.
Circuit 1 contains 4 nets,    Circuit 2 contains 4 nets.

Netlists match uniquely.
Result: Circuits match uniquely.
Logging to file "lvs.log" disabled
LVS Done.
```

Are you using the runme.sh file that was included in the tarball? If not, how are you running netgen?

Thanks, Dan

RTimothyEdwards, 2021-07-10:
Sorry, I missed the tarball up at the top and was just running netgen on the two netlists which you printed out.

If you create your own non-default setup file, then you immediately override the defaults, which includes the default behavior of allowing MOSFET source/drain to permute.  I am not exactly sure why this is so because I thought the behavior was to retain default behavior for permutation unless the setup file specifically used the "permute" command.  Regardless, every setup file should start with "permute default" and "property default" and then start setting any non-default behavior.

mooredan, 2021-07-11:
Yes, adding "permute default" and "property default" at the beginning of the setup.tcl file got this working. Actually, only the former command was needed to get this particular test case working.

Looked in tcltk/netgen.tcl and these are added only when there isn't a setup file specified or present.  Adding them looks to be an easy change. Alternately, a Warning message could be issued if they aren't in the setup file.  Which do you prefer?

mooredan, 2021-07-25:
Created two pull requests to choose from on how to address this issue, if at all. Only one should be merged.