RTimothyEdwards/netgen #23: segfault with caravel/chip_io and flattened primitives (example cells) in magic.

d-m-bailey, 2021-07-14:
As per the conversation on slack.
https://skywater-pdk.slack.com/archives/C017UA7LEUV/p1626212742054500
[netgen-debug-20210714.tar.gz](https://github.com/RTimothyEdwards/netgen/files/6819263/netgen-debug-20210714.tar.gz)

RTimothyEdwards, 2021-07-15:
@d-m-bailey :  Took much longer than I expected, but it was also a worse problem than I expected, failing to set a variable and then more or less randomly overwriting memory.  Anyway, I just fixed it in `commit a332c235248e8d7125d50ef82d1bb8a8fe6aad44` on opencircuitdesign.com.  Because of the error, I am not entirely sure that series resistor addition is correct;  if you see any issues around the devices "respu1" and "respu2" in "gpiov2_pupredrvr_strong_nd2" (W=0.33um, L=11um and 4um, respectively, so the total should be L=15um), let me know.

d-m-bailey, 2023-05-17:
Closing as stale.

closed 2023-05-17 by d-m-bailey