RTimothyEdwards/netgen #28: Latest netgen tool hanging

dineshannayya, 2021-08-08:
I see latest netgen tool hanging after below messages

<pre>Circuits match with 11 symmetries.
Resolving automorphisms by property value.
Resolving automorphisms by pin name.
Netlists match with 9 symmetries.
</pre> 

Last working netgen checkout is : 7ee50a3f8fc41dffad74adec7d32a691586351cf
commit 7ee50a3f8fc41dffad74adec7d32a691586351cf
Author: Tim Edwards <tim@opencircuitdesign.com>
Date:   Mon Mar 1 16:33:55 2021 -0500

Attached the example directory
Run command: run_cmd
 
[exp6.tar.gz](https://github.com/RTimothyEdwards/netgen/files/6950510/exp6.tar.gz)

RTimothyEdwards, 2021-08-09:
I'm not sure what you mean by "latest netgen".  The commit you referenced is dated March 1, which is 30 commits behind the current version.

Nevertheless, the example does demonstrate the issue on the current version of netgen.  I have never seen an example before where the symmetry breaking failed to reduce the number of symmetries.  I will need to investigate.  Thanks for preparing the repeatable test case.

dineshannayya, 2021-08-09:
I mean, last working netgen version for this example is checkout: 7ee50a3f8fc41dffad74adec7d32a691586351cf

RTimothyEdwards, 2021-08-09:
Sorry, I misunderstood.  That makes perfect sense.  On the next commit, I removed the "fast symmetry breaking" method and replaced it with something else.  The fast symmetry breaking worked well for synthesized digital circuits, but it was not guaranteed to work correctly.  Since it is in principle impossible for the new symmetry breaking method to do what it's doing in your example, it is presumably an unexpected bug, and hopefully a simple one.

RTimothyEdwards, 2021-08-09:
Netgen is not hung;  it is just taking a very, very long time to break symmetry because the standard cells were not allowed to combine in parallel.  Your underlying problem is that you do not have the most recent version of open_pdks and the corresponding netgen setup file, which contains the code that allows parallelization of standard cells.

dineshannayya, 2021-08-14:
Hi Timothy

I have tried with latest OpenPDK :

commit 6de165cc41b9f1050006d37288c8121c23756406 (*HEAD -> **master*,
*tag: 1.0.205*, *origin/open_pdks-1.0*, *origin/master*,
*origin/HEAD*)
Author: Tim Edwards ***@***.***>
Date:   Sun Aug 8 20:45:43 2021 -0400

    Modified VERSION to go along with the merge of pull request #142
    from Max Chen.

With Netgen version:

commit e773739e7d41fb645a93069624c7685115f52451 (*HEAD -> **master*,
*origin/master*, *origin/HEAD*)
Author: Tim Edwards ***@***.***>
Date:   Fri Aug 6 11:37:29 2021 -0400

    Updated VERSION as a forced change to check continuous integration
    on github.

In the Run, I see there are series of format issue

Magic 8.2 revision 157 - Compiled on Thu, 07 Nov 2019 21:14:45 +0100.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
/home/dinesha/workarea/opencore/pdk/sky130A/libs.tech/magic/sky130A.tech:
line 19: section version:
	Badly formed version line
Usage: {version text}|{description text}
/home/dinesha/workarea/opencore/pdk/sky130A/libs.tech/magic/sky130A.tech:
line 608: section compose:
	Unrecognized layer (type) name "obsli,obsm1"
/home/dinesha/workarea/opencore/pdk/sky130A/libs.tech/magic/sky130A.tech:
line 796: section cifoutput:
	Unknown statement "bridge".
/home/dinesha/workarea/opencore/pdk/sky130A/libs.tech/magic/sky130A.tech:
line 802: section cifoutput:
	Unknown statement "close".
/home/dinesha/workarea/opencore/pdk/sky130A/libs.tech/magic/sky130A.tech:
line 803: section cifoutput:

I have attached the example and sky130A.tech for your reference.

Not sure if there is any setup issue.  My older netgen setup still works.

-Dinesh.A