RTimothyEdwards/netgen #34: Non symmetric processing

d-m-bailey, 2021-10-15:
@RTimothyEdwards 

In `tcltk/netgen.tcl.in` at line 563 
```
               if {([lsearch $noflat [lindex $endval 0]] == -1) &&
                        ([lsearch $noflat [lindex $endval 1]] == -1)} {
                   netgen::log put "  Flattening non-matched subcircuits $endval\n\n"
                   netgen::flatten class "[lindex $endval 0] $fnum1"
                   netgen::flatten class "[lindex $endval 1] $fnum2"
               } else {
                   netgen::log put "  Continuing with black-boxed subcircuits $endval\n\n"
                   lappend matcherr [lindex $endval 0]
                   # Match pins
```
but at line 612
```
            if {([lsearch $noflat [lindex $endval *1*]] == -1) &&
                    ([lsearch $noflat [lindex $endval 1]] == -1)} {
               netgen::log put "  Flattening non-matched subcircuits $endval\n\n"
               netgen::flatten class "[lindex $endval 0] $fnum1"
               netgen::flatten class "[lindex $endval 1] $fnum2"
            } else {
               netgen::log put "  Continuing with black-boxed subcircuits $endval\n"
               lappend matcherr [lindex $endval 0]
??? duplicate ???
               netgen::log put "  Continuing with black-boxed subcircuits $endval\n"
               lappend matcherr [lindex $endval 0]
               # Match pins
```
The `*1*` should be a `0`, correct? (I added the `*` for emphasis, they're not in the code).
Also are the lines after `??? duplicate ???` correct?

What do you think about appending the file number to `matcherr`? (at both places)

Suggested fixes for code line 612
```
            if {([lsearch $noflat [lindex $endval 0]] == -1) &&
                    ([lsearch $noflat [lindex $endval 1]] == -1)} {
               netgen::log put "  Flattening non-matched subcircuits $endval\n\n"
               netgen::flatten class "[lindex $endval 0] $fnum1"
               netgen::flatten class "[lindex $endval 1] $fnum2"
            } else {
               netgen::log put "  Continuing with black-boxed subcircuits $endval\n"
               lappend matcherr [lindex $endval 0]"(1)"
               lappend matcherr [lindex $endval 1]"(2)"
               # Match pins
```
I'm looking into some other cosmetic changes to this program and can include these if that's ok.

RTimothyEdwards, 2021-10-15:
All good observations and suggestions.  I just implemented all of them and pushed an update.

d-m-bailey, 2022-01-21:
Not specifically tested, but these changes appear to be working. May suggest addition of file numbers to other messages later.

closed 2022-01-21 by d-m-bailey