RTimothyEdwards/netgen #35: segfaults when running device level LVS on caravel's `chip_io` block with these changes.

d-m-bailey, 2021-10-22:
@RTimothyEdwards I get segfaults when running device level LVS on caravel's `chip_io` block with these changes `1.5.205`.

Looks like the netlist pointers are getting whacked. Still haven't quite wrapped my head around the netlist structure, but it looks like it's mainly ports first followed by intermixed nodes, instance pin connections, and parameter records.

As you are no doubt aware, the hash keys for the subcircuits depend on the first port. If the first port changes, pointers throughout the entire netlist need to be reset. Also, if all ports and pins were to be deleted in a circuit, it looks like there's a special case where the port is kept.

What about creating a dummy anchor port that never get processed for each circuit? It doesn't look like there's a node 0, maybe that could be the key. This would keep the hash for each circuit constant and avoid a lot of unnecessary processing, I think.
The program changes required would be to add one line after the start of each loop.
```
assert(tc->cell && IsPort(tc->cell) && tc->cell->node == 0);  // sanity check
for ( ob = tc->cell; !ob ; ob->next ) {
   ...  // last pointer processing
   if ( IsPort(ob) && ob->node == 0 ) continue;  // skip anchor node
```

_Originally posted by @d-m-bailey in https://github.com/RTimothyEdwards/netgen/issues/33#issuecomment-949188663_

RTimothyEdwards, 2021-10-23:
In addition to reverting the code base, there's a further issue here;  the reason that the behavior was changed to flag an error for an unconnected pin was due to an issue that caused us to almost tape out a chip with a missing wire (never mind that the chip had worse problems we didn't know about).  The problem was that the core digital circuit place & route had failed to route a wire out to one of the pins.  It was a pin that connects to the user project area which was empty when running LVS of the digital core, so the pin was not connected to anything, and the behavior caused netgen to say that the circuits matched even though a wire out to a pin was missing.  So while I think the behavior before that was implemented was more or less what you're aiming for, and is the behavior that I prefer, there also needs to be a way to definitively determine that a pin on a subcircuit is connected to the rest of the net.  I think this should be doable.  Meanwhile, to implement what you were trying to do, take a look at the state of the code at revision (and git tag) 1.5.193.  The push of revision 1.5.194 is what started this problem of flagging unconnected pins as errors.

d-m-bailey, 2022-01-21:
Backing out the change eliminated the segfault issue.

closed 2022-01-21 by d-m-bailey