RTimothyEdwards/netgen #38: Can't change column width in lvs report

d-m-bailey, 2021-11-11:
The documentation says that `::netgen::format <col_width>` can be used to change the column width of the lvs report, but results in an error in version `1.5.206`.

```
bash-4.2$ netgen -noc
Netgen 1.5.206 compiled on Thu Oct 28 07:12:42 UTC 2021
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
% ::netgen::format 
41 41
% ::netgen::format 100
command returned bad code: -762612944
% ::netgen::format 
100 100
% quit

bash-4.2$ cat test.source 
puts [netgen::format]
netgen::format 100
puts [netgen::format]

bash-4.2$ netgen -batch source test.source 
Netgen 1.5.206 compiled on Thu Oct 28 07:12:42 UTC 2021
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
41 41
couldn't open "/usr/lib64/tcl8.5/tclIndex": no such file or directory
    while executing
"open [file join $dir tclIndex]"

bash-4.2$ ls /usr/lib64/tcl8.5/        
tclx8.4
bash-4.2$ find /usr/lib64/tcl8.5/ -name tclIndex
bash-4.2$ 
```

The above is from docker, but had the same result in native OS.
```
ciic-cvc:~/mpw-2/caravel-lvs/openlane$ netgen -batch source test.source 
Netgen 1.5.206 compiled on Tue Nov  9 22:10:06 PST 2021
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
41 41
couldn't open "/usr/share/tcltk/tclIndex": no such file or directory
    while executing
"open [file join $dir tclIndex]"

ciic-cvc:~/mpw-2/caravel-lvs/openlane$ ls /usr/share/tcltk
tcl8.6  tk8.6
ciic-cvc:~/mpw-2/caravel-lvs/openlane$ find /usr/share/tcltk -name tclIndex
/usr/share/tcltk/tk8.6/tclIndex
/usr/share/tcltk/tcl8.6/tclIndex
```

It appears that the error occurs during the following call in `_netcmp_format` but may have something to do with the path settings.
`if (Tcl_GetIntFromObj(interp, objv[1], &col1_width) != TCL_OK)`

I tried `export TcLLibPath=/usr/share/tcltk/tcl8.6/` but didn't make a difference.

So batch mode, the error stops execution, but in interactive mode, you can ignore the error and continue it seems.

RTimothyEdwards, 2021-11-11:
I use the environment variable NETGEN_COLUMNS, but I'll look into the error.

RTimothyEdwards, 2021-11-11:
Okay, good catch;  failure to return TCL_OK (or any value) in the (objc > 1) case.  Fixing now.

d-m-bailey, 2022-01-21:
Verified fixed in 1.5.217

closed 2022-01-21 by d-m-bailey