RTimothyEdwards/netgen #40: Pin matching

d-m-bailey, 2021-12-10:
Observations:
- The `MatchPins` is only called if the circuit topology matches.
- Only pin matches only show pin information. Pin matches to nets and disconnected pins are indistinguishable.
- proxy pins work in some situations but cause mismatches in others.

RTimothyEdwards, 2021-12-11:
Intuitively, it seems like you should be able to match pins on a mismatching topology.  In practice, I'm not sure how easy that would be.  It might be possible to scan the matched 1:1 partitions for pins and dump those results and mark all of the pins that show up in illegal partitions as "unresolved" or something like that.

d-m-bailey, 2021-12-11:
Currently, I'm printing the port correspondence of unmatched subcircuits for the top subcircuit only. Lower cell processing is the same as before - port correspondence is only printed for topologically matching circuits.

d-m-bailey, 2022-01-21:
Moved to discussions

closed 2022-01-21 by d-m-bailey