RTimothyEdwards/netgen #53: segfault when combing parallel devices

d-m-bailey, 2022-01-21:
Netgen 1.5.217

This is for the fully flattened `sky130_ef_io__gpiov2_pad_wrapped` cell from caravel `chip_io`.
```
Property l in circuit1 has no matching property in circuit2
sky130_fd_pr__res_generic_m1:181 vs. sky130_ef_io__gpiov2_pad:gpiov2_ef_q0/sky130_fd_io__top_gpiov2:gpiov2_base_q0/sky130_fd_io__gpiov2_opath:opath_q0/sky130_fd_io__gpiov2_octl_dat:opath_q0/sky130_fd_io__gpiov2_obpredrvr:predrvr_q0/sky130_fd_io__gpiov2_pdpredrvr_strong:pd_strong_q0/sky130_fd_io__tk_opto:I77/sky130_fd_io__tk_em1s:e2_q0/sky130_fd_pr__res_generic_m1:I1:
Property w in circuit1 has no matching property in circuit2
Segmentation fault (core dumped)
```
To duplicate:
1. `tar xzf merge-issue.tar.gz`
2. `cd merge-issue`
3. `netgen -batch source setup_file.gds.lvs`

[merge-issue.tar.gz](https://github.com/RTimothyEdwards/netgen/files/7914777/merge-issue.tar.gz)

RTimothyEdwards, 2022-01-22:
I'm on 1.5.219, which includes a modification in the parallel combination code as suggested by Anton Blanchard, and I didn't get a segfault from that example.  Can you confirm that?

d-m-bailey, 2022-01-22:
1.5.219 works for me too! Sorry for the wasted effort.

closed 2022-01-22 by d-m-bailey

LaurentChr, 2023-10-03:
I got segmentation fault error with one of my netlist : hard to debug ... the error was : on a MOS parameter : L=1..5u
This double point generated the Segmentation fault error.

Laurent

RTimothyEdwards, 2023-10-03:
@LaurentChr :  This should be a new issue, as I'm sure it has nothing to do with the original post---but then I could be wrong.  I did try just changing a random netlist from "L=0.15" to "L=0..15", and I consistently get output like this:
```
 l circuit1: "0.15"   circuit2: "0..3"   (exact match req'd)
```
In this case, netgen looks like it's doing the right thing.  It does not recognize one entry as a number, so it treats it as a string and then requires an exact string match against the other netlist.  That's enough to bring the syntax error to my attention.

It's possible that there are issues when the device with the syntax error is combined with a parallel device.  If you can provide an example netlist that causes a segfault, it will be helpful.  Otherwise I'll poke around and see if I can make it crash.

RTimothyEdwards, 2023-10-03:
Ooh, never mind.  It does have something to do with combining parallel devices (although still different from the original post).  I have a failing example and will use that to debug.

RTimothyEdwards, 2023-10-03:
@LaurentChr :  Fixed in version 1.5.259.  One spot in the code failed to check types of both properties when comparing two properties, leading to a segfault when a float or integer is incorrectly interpreted as a string.