RTimothyEdwards/netgen #54: Order of verilog files effects LVS results

d-m-bailey, 2022-03-17:
It appears that the order of the verilog files effects the connectivity.

If the parent verilog is read before the child verilog, a placeholder is created. When the actual verilog is input, the ports do not align correctly. I'll attach a minimal test case later, but here are the results of a comparison. 

```
  % nodes powergood_check mgmt_protect                                                               |  % nodes powergood_check mgmt_protect                                                              
  Circuit 'mgmt_protect' not found.                                                                  |  Circuit 'mgmt_protect' not found.                                                                 
  Device 'powergood_check' Pins:                                                                     |  Device 'powergood_check' Pins:                                                                    
     ...                                                                                             |     ...                                                                                            
     Pin 3 (mprj2_logic_high_hvl/VGND) = vssa1 (port of mgmt_protect)                                |     Pin 3 (mprj2_logic_high_hvl/VGND) = vssa2 (port of mgmt_protect)                               
     Pin 4 (mprj2_logic_high_hvl/VNB) = vssa1 (port of mgmt_protect)                                 |     Pin 4 (mprj2_logic_high_hvl/VNB) = vssa2 (port of mgmt_protect)                                
     Pin 5 (mprj2_logic_high_hvl/VPWR) = vdda1 (port of mgmt_protect)                                |     Pin 5 (mprj2_logic_high_hvl/VPWR) = vdda2 (port of mgmt_protect)                               
     Pin 6 (mprj2_logic_high_hvl/VPB) = vdda1 (port of mgmt_protect)                                 |     Pin 6 (mprj2_logic_high_hvl/VPB) = vdda2 (port of mgmt_protect)                                
     ...                                                                                             |     ...                                                                                            
     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/3) = vssa2 (port of mgmt_protect)     |     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/3) = vssd (port of mgmt_protect)     
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/4) = vssa2 (port of mgmt_protect)     |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/4) = vssd (port of mgmt_protect)     
     ...                                                                                             |     ...                                                                                            
     Pin 1 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/1) = vdda1 (port of mgmt_protect)     |     Pin 1 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/1) = vdda2 (port of mgmt_protect)    
     ...                                                                                             |     ...                                                                                            
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/4) = vdda1 (port of mgmt_protect)     |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/4) = vdda2 (port of mgmt_protect)    
     ...                                                                                             |     ...                                                                                            
     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_01v8:2/3) = vssa2 (port of mgmt_protect)          |     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_01v8:2/3) = vssd (port of mgmt_protect)          
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_01v8:2/4) = vssa2 (port of mgmt_protect)          |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_01v8:2/4) = vssd (port of mgmt_protect)          
     ...                                                                                             |     ...                                                                                            
     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:3/3) = vssa2 (port of mgmt_protect)     |     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:3/3) = vssd (port of mgmt_protect)     
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:3/4) = vssa2 (port of mgmt_protect)     |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:3/4) = vssd (port of mgmt_protect)     
     ...                                                                                             |     ...                                                                                            
     Pin 1 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:7/1) = vssa2 (port of mgmt_protect)     |     Pin 1 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:7/1) = vssd (port of mgmt_protect)     
     ...                                                                                             |     ...                                                                                            
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:7/4) = vssa2 (port of mgmt_protect)     |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:7/4) = vssd (port of mgmt_protect)     
     Pin 1 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:10/1) = vdda1 (port of mgmt_protect)    |     Pin 1 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:10/1) = vdda2 (port of mgmt_protect)   
     ...                                                                                             |     ...                                                                                            
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:10/4) = vdda1 (port of mgmt_protect)    |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:10/4) = vdda2 (port of mgmt_protect)   
     ...                                                                                             |     ...                                                                                            
     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:11/3) = vssa2 (port of mgmt_protect)    |     Pin 3 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:11/3) = vssd (port of mgmt_protect)    
     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:11/4) = vssa2 (port of mgmt_protect)    |     Pin 4 (mprj2_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:11/4) = vssd (port of mgmt_protect)    
     ...                                                                                             |     ...
     Pin 3 (mprj_logic_high_hvl/VGND) = vdda2 (port of mgmt_protect)                                 |     Pin 3 (mprj_logic_high_hvl/VGND) = vssa1 (port of mgmt_protect)                                
     Pin 4 (mprj_logic_high_hvl/VNB) = vdda2 (port of mgmt_protect)                                  |     Pin 4 (mprj_logic_high_hvl/VNB) = vssa1 (port of mgmt_protect)                                 
     Pin 5 (mprj_logic_high_hvl/VPWR) = vssd (port of mgmt_protect)                                  |     Pin 5 (mprj_logic_high_hvl/VPWR) = vdda1 (port of mgmt_protect)                                
     Pin 6 (mprj_logic_high_hvl/VPB) = vssd (port of mgmt_protect)                                   |     Pin 6 (mprj_logic_high_hvl/VPB) = vdda1 (port of mgmt_protect)                                 
     ...                                                                                             |     ...
     Pin 3 (mprj_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/3) = vssa2 (port of mgmt_protect)      |     Pin 3 (mprj_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/3) = vssd (port of mgmt_protect)      
     Pin 4 (mprj_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/4) = vssa2 (port of mgmt_protect)      |     Pin 4 (mprj_logic_high_lv/sky130_fd_pr__nfet_g5v0d10v5:0/4) = vssd (port of mgmt_protect)      
     ...                                                                                             |     ...
     Pin 1 (mprj_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/1) = vssd (port of mgmt_protect)       |     Pin 1 (mprj_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/1) = vdda1 (port of mgmt_protect)     
[... cut ...]
     ...                                                                                             |     ...                                                                                            
debug.ng [+]                                                                       32,6           Top debug.ok [+]                                                                      32,6           Top
```
The incorrect results are on the left and the correct results are on the right.
Particularly notice the pfet bulk connection to vssd. 
`Pin 4 (mprj_logic_high_lv/sky130_fd_pr__pfet_g5v0d10v5:1/4) = vssd (port of mgmt_protect)`

d-m-bailey, 2022-03-17:
The work around is, of course, to read verilog child modules before they are referenced.

Problems can be detected with `grep 'Creating placeholder cell definition for module'` on the output log (NOT the lvs log).

RTimothyEdwards, 2024-02-18:
@d-m-bailey :  I believe that netgen version 1.5.270 fixes this issue.  I used the same approach (pretty much the same code) that I used to deal with the same problem of order of input files when mixing verilog and SPICE.  Appears to work correctly on the example that Kareem Farid posted recently.