RTimothyEdwards/netgen #58: noflatten option does not work for instance count mismatches

d-m-bailey, 2022-09-07:
Specifying cells with the `-noflatten` option should prevent those cells from being flattened.

However, if there is a difference in the cell counts, netgen will flatten cells regardless of whether or not they're tagged `noflatten`.

`Netgen 1.5.227`

From the stdout:
```
Will not flatten these subcells: sky130_fd_sc_hd__decap_12 sky130_fd_sc_hd__decap_3 sky130_fd_sc_hd__decap_4 sky130_fd_sc_hd__decap_6 sky130_fd_sc_hd__decap_8 sky130_fd_sc_hd__diode_2
```

From the lvs report:
```
Flattening instances of sky130_fd_sc_hd__diode_2 in cell uP16_efabless (0) makes a better match
Flattening instances of sky130_fd_sc_hd__diode_2 in cell uP16_efabless (1) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_12 in cell uP16_efabless (0) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_12 in cell uP16_efabless (1) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_3 in cell uP16_efabless (0) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_3 in cell uP16_efabless (1) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_4 in cell uP16_efabless (0) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_4 in cell uP16_efabless (1) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_6 in cell uP16_efabless (0) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_6 in cell uP16_efabless (1) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_8 in cell uP16_efabless (0) makes a better match
Flattening instances of sky130_fd_sc_hd__decap_8 in cell uP16_efabless (1) makes a better match
```

RTimothyEdwards, 2022-09-07:
Should be simple enough to fix. . .

RTimothyEdwards, 2022-09-07:
Okay, straightforward but not simple.  The `-noflatten` option only works on the script level and has not correspondence in the netgen code.  I will need to implement some command line option like `flatten <cell> prohibit` and build up a list of cells that can be checked against during prematch or after a failed match to prohibit flattening.

RTimothyEdwards, 2022-09-07:
Okay, I am implementing what I mentioned above (almost):  `flatten prohibit <cell>` or `flatten deny <cell>` will mark a cell as a placeholder, which will prevent it from being flattened.  The command option will be called for each cell specified by the `-noflatten` option, although the command can also be put in the setup file (which would render the `-noflatten` option unnecessary).

I have coded this up but I need to test it because I have questions around the use of the placeholder flag vs. setting the cell class to type "module";  these two are related but I forget the circumstances under which each is used.  I'll post an update tomorrow (ping me if I don't).

RTimothyEdwards, 2022-09-12:
Fixed in version 1.5.228.   `-noflatten` will now call a new command line option `flatten prohibit <cell_name>` which then prohibits cells from being flattened at any time during a comparison.

I also discovered another bug in which the `-noflatten=<cell_name>` syntax does not work (only works with a file name) and fixed that as well.

d-m-bailey, 2025-10-22:
@RTimothyEdwards thanks for the prompt response. This should make pinpointing errors a bit easier.