RTimothyEdwards/netgen #60: netgen generates json with invalid escape sequence

proppy, 2022-09-17:
It seems that netgen generates invalid escape sequence when run on the `spm` design with the `gf180mcuC` PDK:
```
Logging to file "/content/OpenLane/designs/spm/runs/RUN_2022.09.17_01.33.20/logs/signoff/35-spm.lef.log" disabled
LVS Done.
Traceback (most recent call last):
  File "/content/OpenLane/scripts/count_lvs.py", line 118, in <module>
    failures = count_LVS_failures(args.file)
  File "/content/OpenLane/scripts/count_lvs.py", line 37, in count_LVS_failures
    lvsdata = json.load(cfile)
  File "/content/conda-env/lib/python3.7/json/__init__.py", line 296, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/content/conda-env/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/content/conda-env/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/content/conda-env/lib/python3.7/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Invalid \escape: line 774 column 10 (char 13637)
child process exited abnormally
    while executing
"exec python3 $::env(SCRIPTS_DIR)/count_lvs.py  -f $extraction_prefix.json  |& tee $::env(TERMINAL_OUTPUT) $count_lvs_log"
    (procedure "run_lvs" line 75)
    invoked from within
"run_lvs"
    (procedure "run_lvs_step" line 10)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "OpenLane/flow.tcl" line 401)
```
Looking at the attached JSON: [35-spm.lef.json.gz](https://github.com/RTimothyEdwards/netgen/files/9590773/35-spm.lef.json.gz), it seems that it comes from the following sequence that include a `\` in the cell name:
```
        "\genblk1[7].csa.sc ",
        "\genblk1[8].csa.sc ",
        "\genblk1[8].csa.y ",
        "\genblk1[10].csa.y ",
        "\genblk1[14].csa.y ",
        "\genblk1[15].csa.sc ",
        "\genblk1[22].csa.sc ",
        "\genblk1[29].csa.y ",
        "\genblk1[30].csa.sc ",
        "\genblk1[30].csa.y ",
```
It should instead be escaped with:
```
        "\\genblk1[7].csa.sc ",
        "\\genblk1[8].csa.sc ",
        "\\genblk1[8].csa.y ",
        "\\genblk1[10].csa.y ",
        "\\genblk1[14].csa.y ",
        "\\genblk1[15].csa.sc ",
        "\\genblk1[22].csa.sc ",
        "\\genblk1[29].csa.y ",
        "\\genblk1[30].csa.sc ",
        "\\genblk1[30].csa.y ",
```

proppy, 2022-09-17:
See this notebook that reproduce the issue:
https://colab.sandbox.google.com/gist/proppy/21d9607540800c70c313c78bbebd84bf/gf180mcu-openlane-playground.ipynb

proppy, 2022-09-17:
Looks like things would need to be escaped in the following block:
https://github.com/RTimothyEdwards/netgen/blob/9297090dc1f4896f724c4b5beacdc60aab51df22/tcltk/netgen.tcl.in#L60-L82
I wonder if https://wiki.tcl-lang.org/page/Tcllib+JSON could be used instead?

RTimothyEdwards, 2022-09-17:
@proppy :  (Stupid verilog backslash notation---I've spent hours and hours coding and debugging stuff because of somebody's STUPID decision to implement this harebrained syntax).  What version of verilog?  There is code in netgen that escapes backslashes, but (1) I recoded parts of netgen in the last couple of days to incorporate some code written by Mitch Bailey, and (2) that code might not affect some output that goes into the JSON file (more stupid syntax restrictions I have to meet) anyway.

RTimothyEdwards, 2022-09-17:
@proppy :  In cases of nets I handle the backslash escapes.  Using backslash escapes on pin names is a good way to cause major issues in tools and scripts throughout the flow, so it is really not recommended to do that.  But fixing it in the Tcl script is trivial, so I'll get that done right away.

RTimothyEdwards, 2022-09-17:
@proppy:  Fixed in netgen version 1.5.232.

proppy, 2022-09-21:
verified that the pins get escaped properly and properly parsable:
```
$ python -m json.tool /content/OpenLane/designs/spm/runs/RUN_2022.09.21_01.23.13/logs/signoff/35-spm.lef.json | grep '\\' | head
                "\\genblk1[12].csa.y ",
                "\\genblk1[19].csa.sc ",
                "\\genblk1[21].csa.y ",
                "\\genblk1[24].csa.y ",
                "\\genblk1[29].csa.sc ",
                "\\genblk1[29].csa.y ",
                "\\csa0.hsum2 ",
                "\\genblk1[11].csa.hsum2 ",
                "\\genblk1[13].csa.hsum2 ",
                "\\genblk1[17].csa.hsum2 ",
```

But I do get a LVS error now:
```
LVS reports:
    net count difference = 0
    device count difference = 0
    unmatched nets = 0
    unmatched devices = 0
    unmatched pins = 392
    property failures = 0

Total errors = 392
```

But I wonder if that's related since the number of unmatched pins doesn't to match the number of escaped entries?
```
$ !python -m json.tool /content/OpenLane/designs/spm/runs/RUN_2022.09.21_01.23.13/logs/signoff/35-spm.lef.json | grep '\\' | wc -l
94
```

proppy, 2022-09-21:
closing this (since now pins are properly escaped) and filing https://github.com/The-OpenROAD-Project/OpenLane/issues/1380 to investigate things on the OpenLane side.

closed 2022-09-21 by proppy