RTimothyEdwards/netgen #68: LVS issue when running on gf180mcu

marwaneltoukhy, 2022-11-28:
I'm facing an LVS issue when running on any design technology gf180mcuC. The foundry gds has an extra layer `63/63` which has text related to the cell name/version/metrics, netgen treats those as pins and produces an LVS issue. Not sure if this is a netgen issue or a magic extraction issue, if it's a magic issue, I can move this issue there.

simple design for testing: `mprj_io_buffer` can be found [here](https://github.com/efabless/caravel-gf180mcu/tree/PnR) (branch: PnR)

marwaneltoukhy, 2022-11-28:
@RTimothyEdwards can't assign you as I am not a collaborator on this repo

d-m-bailey, 2022-11-28:
@marwaneltoukhy Haven't run any gf LVS myself, but looking through the open_pdks commits, looks like the `63/63` problem was handled recently. Maybe the pdk your using doesn't have the latest tech.
https://github.com/RTimothyEdwards/open_pdks/commit/a519523b0d9bc913a6f87a5eed083597ed9e2e93

RTimothyEdwards, 2022-11-28:
@marwaneltoukhy :  You need to run netgen on a netlist created from a GDS that has been generated using the latest PDK version from open_pdks.  Just updating the PDK from open_pdks isn't going to solve the problem if the layout being extracted has a GDS file containing cells from an earlier build of the PDK.

marwaneltoukhy, 2022-11-29:
Closed with updating PDK and reharndening - new issue https://github.com/RTimothyEdwards/magic/issues/203

closed 2022-11-29 by marwaneltoukhy