RTimothyEdwards/netgen #69: gf180mcuC lvs mismatch

kareefardi, 2022-12-02:
using latest negten, open_pdks and magic. attached are two testcase that include gatelevel netlist, spice and def/lef used to extract the spice. one testcase has the `VPW` and `VNW` in the gatelevel netlist and the other doesn't. 
extraction method used:
```
extract do local
extract no capacitance
extract no coupling
extract no resistance
extract no adjust
extract

ext2spice lvs
ext2spice -o $::env(EXT_NETLIST) $::env(DESIGN_NAME).ext
```
[issue-vpw-vnw.tar.gz](https://github.com/RTimothyEdwards/netgen/files/10141253/issue-vpw-vnw.tar.gz)
[issue-no-vpw-vnw.tar.gz](https://github.com/RTimothyEdwards/netgen/files/10141274/issue-no-vpw-vnw.tar.gz)

RTimothyEdwards, 2022-12-02:
@kareefardi :  I have to see the layout.  The netlist shows that the wells and substrates are disconnected throughout the design.  But I'm able to extract layouts from caravel-gf180mcu, and I don't see any issues in the layout and extracted netlist.  So whatever is going wrong is happening before LVS.

kareefardi, 2022-12-02:
@RTimothyEdwards the layout is attached in both testcases as a def-lef view.

I have also just noticed that in `issue-no-vpw-vnw.tar.gz`  VDD and VSS are also present in the nets section. Perhaps that's due to missing connections to VNW and VPW in the gatelevel netlist

RTimothyEdwards, 2022-12-02:
@kareefardi :  Ah.  The problem is that the "fill" cells in the GF standard cell library have no nwell, so the nwell is breaking where the fill cells are used.

RTimothyEdwards, 2022-12-02:
This is bizarre.  The GDS view of the fill cells has nwell and pwell, but the .mag view has lost everything except the metal.  This is a PDK issue and should be posted in open_pdks.  I'm investigating.

kareefardi, 2022-12-03:
I believe this is fixed now (by patches to the abstract views of std cells)

closed 2022-12-03 by kareefardi