RTimothyEdwards/netgen #72: SIGSEGV during verilog read for mgmt_core_wrapper

d-m-bailey, 2023-03-04:
See https://github.com/RTimothyEdwards/netgen/commit/67da250615a22a0303fed416aa2c5c90c5b72078#commitcomment-102993180

Here's the gdb trace
```
Program received signal SIGSEGV, Segmentation fault.
__strncpy_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcpy-sse2-unaligned.S:298
298     ../sysdeps/x86_64/multiarch/strcpy-sse2-unaligned.S: No such file or directory.
(gdb) bt
#0  __strncpy_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcpy-sse2-unaligned.S:298
#1  0x00007ffff76152db in ReadVerilogFile (fname=0x555575da1370 "/home/user/mpw-9/caravel_user_project2/mgmt_core_wrapper/verilog/gl/mgmt_core_wrapper.v", filenum=1,
    CellStackPtr=0x7fffffffd470, blackbox=0) at verilog.c:1593
#2  0x00007ffff7617e4e in ReadVerilogTop (fname=0x555575da1370 "/home/user/mpw-9/caravel_user_project2/mgmt_core_wrapper/verilog/gl/mgmt_core_wrapper.v",
    fnum=0x7fffffffd4e8, blackbox=0) at verilog.c:2478
#3  0x00007ffff7617f16 in ReadVerilog (fname=0x555575da1370 "/home/user/mpw-9/caravel_user_project2/mgmt_core_wrapper/verilog/gl/mgmt_core_wrapper.v",
    fnum=0x7fffffffd4e8) at verilog.c:2502
#4  0x00007ffff761b878 in _netgen_readnet (clientData=0x0, interp=0x555555566120, objc=3, objv=0x555555573cf0) at tclnetgen.c:817
#5  0x00007ffff7c1bfb7 in TclNRRunCallbacks () from /lib/x86_64-linux-gnu/libtcl8.6.so
#6  0x00007ffff7c1d3af in ?? () from /lib/x86_64-linux-gnu/libtcl8.6.so
#7  0x00007ffff7cd63f8 in Tcl_FSEvalFileEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#8  0x00007ffff7cd4d57 in Tcl_EvalFile () from /lib/x86_64-linux-gnu/libtcl8.6.so
#9  0x00007ffff7cddf6e in Tcl_SourceRCFile () from /lib/x86_64-linux-gnu/libtcl8.6.so
#10 0x00007ffff7cde4bd in Tcl_MainEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#11 0x0000555555555208 in main (argc=4, argv=0x7fffffffdb78) at netgenexec.c:82
(gdb) l
~      
293     in ../sysdeps/x86_64/multiarch/strcpy-sse2-unaligned.S
(gdb) up
#1  0x00007ffff76152db in ReadVerilogFile (fname=0x555575da1370 "/home/user/mpw-9/caravel_user_project2/mgmt_core_wrapper/verilog/gl/mgmt_core_wrapper.v", filenum=1,
    CellStackPtr=0x7fffffffd470, blackbox=0) at verilog.c:1593
1593                                strncpy(nodename, MAX_STR_LEN, lhs->name);
(gdb) p nodename
$1 = "FILLER_290_5647", '\000' <repeats 240 times>
(gdb) p lhs->name
$2 = 0x555578cfbae0 "debug_out"
(gdb) p MAX_STR_LEN
No symbol "MAX_STR_LEN" in current context.
```
Maybe constants don't print in gdb.

I'll add some debugging commands to check string lengths and report later.

d-m-bailey, 2023-03-04:
`strncpy` parameters in recently changed `base/verilog.c `are in the wrong order.

RTimothyEdwards, 2023-03-04:
Fixed in version 1.5.248 (just pushed to opencircuitdesign.com).  Thanks for debugging!

closed 2023-04-22 by RTimothyEdwards