RTimothyEdwards/netgen #79: Unable to launch netgen

vijayank88, 2023-07-14:
Using Ubuntu 20.04. 
Installed  with `./configure make sudo make install`
```
$ sudo make install
./scripts/mkdirs /usr/local/bin
--- installing executable to /usr/local/bin
--- installing run-time files to /usr/local/lib
```

If I launch `netgen`, it is failing with below log:
```
$ netgen
application-specific initialization failed: Can't find a usable tk.tcl in the following directories: 
    /usr/local/lib/tcl8.6/tk8.6 /usr/local/lib/tk8.6 /usr/lib/tk8.6 /usr/lib/tk8.6 /lib/tk8.6 /usr/library

This probably means that tk wasn't installed properly.

Error in startup script: invalid command name "::tk::unsupported::ExposePrivateVariable"
    while executing
"::tk::unsupported::ExposePrivateVariable tkPriv"
    invoked from within
"if {![llength [info globals tkPriv]]} {
    ::tk::unsupported::ExposePrivateVariable tkPriv
}"
    (file "/usr/local/lib/netgen/tcl/tkcon.tcl" line 65)
```
I tried to install `tcl-dev` and `tk-dev` it says both or exist.
```
tcl8.6-dev is already the newest version (8.6.10+dfsg-1).
tk8.6-dev is already the newest version (8.6.10-1).
```
How to resolve that error.

RTimothyEdwards, 2023-07-14:
If you tried a build before installing tcl-dev and tk-dev, then you might have to do a `make clean` and start over again with `./configure`.  Otherwise you might end up with the system trying to link together files that were compiled with Tcl/Tk support and other files that weren't.  (FYI, I don't think this is a version issue because I'm using Tk version 8.6.10-7 and the Tk console which generated the error message has been working for years).

Otherwise, the error message seems to be Tk-specific (not Tcl).  You may still be able to run in terminal mode (`netgen -noconsole`) or batch mode (`netgen -batch ...`).

RTimothyEdwards, 2023-07-14:
Note that in practice, netgen is almost always used in batch mode (at least that's the way I always use it).

vijayank88, 2023-07-15:
Thanks @RTimothyEdwards.
It is working with `-noconsole`
```
~$ netgen -noconsole
Netgen 1.5.255 compiled on Friday 14 July 2023 04:52:36 PM IST
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
% 
```

closed 2023-07-15 by vijayank88