RTimothyEdwards/netgen #84: Using netgen to generate ".sim" files from ".spice" netlists

mhajimorad, 2023-10-26:
Hi,

I have created a circuit schematic in xschem using the sky130 PDK, and am trying to use netgen to generate a ".sim" file from xschem's created ".spice file".  The intended purpose of the ".sim" file is to allow my students to use IRSIM to verify the switching behavior of transistor-based schematics.

Upon launching netgen, I have tried to open xschem's created ".spice" file using commands like
`readnet spice my_netlist.spice`

But I obtain messages like `call to undefined subcircuit sky130...`

And when I try to include the path to the spice models in the ".spice" netlist file itself, netgen gives a large number of errors.

Any tips/suggestions on how I can use netgen to generate ".sim" files from xschem produced sky130 spice netlist files?

Thank you!

RTimothyEdwards, 2023-10-26:
Netgen's ability to switch between formats is rather far behind in development at this point.  The problem here is that to deal with multiple types of transistors and devices declared as subcircuits, I switched from using "n" and "p" records in the .sim file to using "x" records.  However, the "x" records still refer to devices of a specific type like nFETs and pFETs, and have a pin order that is specific to the .sim file.  It is not possible to know what kind of a device is represented by "x" without a definition somewhere;  that definition is given in the ".prm" files for IRSIM in the PDK.  But that file and file format is unknown to netgen.  I would need to write an extension to netgen;  essentially a separate read routine to read a .prm file and use it to define all of the subcircuit types, and then it would know how to write out the .sim file.  I don't think that would be particularly hard to do.

mhajimorad, 2023-10-26:
@RTimothyEdwards , thank you for your message!

Do you know by chance whether the extension you had referenced is something that can be done sometime in the near future?  I simply ask because it will greatly help with the planning of the curriculum/labs of our introductory VLSI course.

Thank you!

RTimothyEdwards, 2023-10-26:
I'm trying to figure out when I can get it done.  When do you need it by?

mhajimorad, 2023-10-26:
@RTimothyEdwards ,

If possible, in the next month or so?  I was hoping to learn the different steps involved in getting xschem/netgen/irsim to communicate with each other by Thanksgiving at the latest, so that I can have students work with these tools after Thanksgiving.

mhajimorad, 2023-10-26:
@RTimothyEdwards ,

Thank you so much for the quick update!

Upon installing the updated netgen from opencircuitdesign.com , I receive the following message after running your first command in netgen.

More specifically, after entering
`set filenum [readnet prm /usr/local/share/pdk/sky130A/libs.tech/irsim/sky130A_tt_nom_27.prm]`

I receive the message
`bad format "prm": must be automatic, ext, extflat, sim, ntk, spice, verilog, netgen, actel, or xilinx`

RTimothyEdwards, 2023-10-26:
"on opencircuitdesign" means you need to do a git clone from the site, not download the tarball.  The updated version is 1.5.261.  (IRSIM version is 9.7.118).

RTimothyEdwards, 2023-10-26:
The tarball is automatically generated at the same time as the github mirror is updated, which happens overnight.

mhajimorad, 2023-10-26:
@RTimothyEdwards ,

Thank you!  I was able to download the updated versions.

So far, I have tried to test a simple CMOS inverter implemented in sky130. The generated .SIM file reads as follows:

```
| units: 20000	tech: scmos
x OUT IN GND GND l=30 w=200 x=0 y=0 sky130_fd_pr__nfet_01v8
x OUT IN VDD VDD l=30 w=200 x=0 y=0 sky130_fd_pr__pfet_01v8
```

Upon launching IRSIM (loading prm file in the process) and typing the following sequence continually yields **undefined** for node **OUT**:

```
% h VDD
% l GND
% l IN
% watchnode OUT
% s
% h IN
% s
% l IN
% s
```

Is there something that I am doing wrong?

RTimothyEdwards, 2023-10-27:
@mhajimorad :  I am going to have to rethink something here. . . the pins are being output in the order they were read from the SPICE file, but .sim files use order G-S-D-B for FETs while SPICE uses order D-G-S-B.  I will need to fix that.

RTimothyEdwards, 2023-10-27:
@mhajimorad :  It should be fixed now.  Thanks for the feedback!

mhajimorad, 2023-10-27:
@RTimothyEdwards ,

Thank you!

So that I can also test/confirm some xschem schematics that contain multiple levels of hierarchy (which lead to SPICE netlists with multiple levels of SUBCKT definitions), you had mentioned something about "flattening" in netgen.

To flatten a SPICE netlist, are there sequences of netgen commands you recommend to be entered?  I assume these would go after the `readnet` and before the `writenet` commands?

Thank you again!

RTimothyEdwards, 2023-10-27:
@mhajimorad :  I had to push another correction to netgen to get the flattening to work correctly (netgen version 1.5.262).
With the update, here's a simple test using the Efabless caravel repository from https://github.com/efabless/caravel.  After using xschem to write the netlist of `simple_por.sch` into `simple_por.sim`, run `netgen` and issue the following commands:
```
set filenum [readnet prm $PDK_ROOT/sky130A/libs.tech/irsim/sky130A_tt_nom_27.prm]
readnet spice $PDK_ROOT/sky130A/libs.ref/sky130_fd_sc_hvl/spice/sky130_fd_sc_hvl.spice $filenum
readnet spice simple_por.spice $filenum
flatten simple_por
writenet sim simple_por
```
This is probably not a great example to use because it is an analog circuit and is not simulatable in IRSIM, but hopefully that shows several things, such as how to read in netlists from the PDK library (for the several standard cells instantiated in the schematic) and how to flatten the design and write the flattened netlist out to .sim format.

mhajimorad, 2023-10-27:
@RTimothyEdwards ,

Thank you for your message.

I did try an example from xschem's "xschemrc" file, where the circuit contains CMOS NAND and NOT circuits drawn in a hierarchical fashion.  I have attached the SPICE file that xschem produces.

I subsequently used the latest version of netgen to produce a SIM file, using the following commands in the process
```
set filenum [readnet prm $PDK_ROOT/sky130A/libs.tech/irsim/sky130A_tt_nom_27.prm]
readnet spice my_netlist.spice $filenum
flatten my_netlist.spice
writenet sim my_netlist.spice
```

I did receive subckt undefined messages after the second line, but I am guessing this is why we subsequently run the "flatten" command afterward on the third line.  In case it is of interest, I have also attached the produced SIM file.  Upon testing the SIM file in IRSIM, the circuit appears to correctly simulate.  

Before testing additional example circuits that have more levels of hierarchy, I just wanted to confirm that things look okay with the SIM file.  For instance, there does not appear to be the same number of pins for the NAND gate in the flattened SIM file as there are in the SPICE SUBCKT definition.  Is this to be expected?

Thank you for all of your help!
 
[sky130_cmos_nand_inverter.spice.txt](https://github.com/RTimothyEdwards/netgen/files/13192493/sky130_cmos_nand_inverter.spice.txt)
[sky130_cmos_nand_inverter.sim.txt](https://github.com/RTimothyEdwards/netgen/files/13192500/sky130_cmos_nand_inverter.sim.txt)

RTimothyEdwards, 2023-10-28:
I'm not sure what you mean by the .sim

RTimothyEdwards, 2023-10-28:
I have gotten rid of most of the "undefined" messages, but there are a few device types that are not in the PRM file.  I was thinking that probably there is a better way to handle the device types through the setup file instead of reading the PRM file, but it's something I'll need to code.

Your example files look good.  Let me know if you have any other issues.

mhajimorad, 2023-11-01:
@RTimothyEdwards ,

Thank you again for all of your help in getting `netgen` to generate sim files from sky130-based spice netlists.  So far, my simple tests have been working without issue.

Out of curiosity, is the `spice --> netgen --> irsim` pipeline meant to also work when a `spice` netlist contains sky130 digital standard cell definitions?  I tried to use the pipeline for a carry-look-ahead adder example that is provided in `xschem`, where the constituent building blocks are sky130 digital standard cells.  Both `netgen` and `irsim` produce warnings/errors (mostly originating from the standard cells being undefined).

In case it is of interest, I have attached the `spice` netlist + `netgen`-produced sim files.
[hierarchy2.spice.txt](https://github.com/RTimothyEdwards/netgen/files/13230826/hierarchy2.spice.txt)
[hierarchy2.sim.txt](https://github.com/RTimothyEdwards/netgen/files/13230832/hierarchy2.sim.txt)

Thank you again for all of your help!

RTimothyEdwards, 2023-11-01:
@mhajimorad :  This should work if you do `readnet spice $PDK_ROOT/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice $filenum` in between reading the .prm file and the main netlist.

mhajimorad, 2023-11-01:
Thank you @RTimothyEdwards !

In IRSIM, is there specific syntax that should be used to access pins labeled as A[0], A[1], ... , A[31]?

For instance, when I enter the IRSIM command `h A[0]` , I get an `invalid command name "0"` error?

And for the standard cells, how does one specify high and low for the power and ground nets?  When I enter IRSIM commands `h VCC` and `l VSS` , I similarly get `no such node or vector` errors.

Thank you!

mhajimorad, 2023-11-01:
@RTimothyEdwards ,

Please ignore my previous post.  I think I may have been able to resolve the issues.

Thank you again for everything!

RTimothyEdwards, 2023-11-02:
Okay.  For the bracket notation, brackets are operators in Tcl, so they have to be escaped, as in `h A\[0\]`.

mhajimorad, 2023-11-17:
@RTimothyEdwards ,

Thank you again for all of your assistance.  My additional testing has gone well, so I will be go ahead and release the material to students in my introductory VLSI course.  I will let you know if any issues arise as they use the tools for their assignments/project (but it could take a while because of the learning curve associated with them getting comfortable with the material + using the tools).

Thank you again for everything!

mhajimorad, 2023-11-29:
@RTimothyEdwards ,

I hope all is well.  It didn't take long for one of my students to discover an issue.  Upon trying to use netgen to convert the attached spice netlist to an irsim sim file, we get an empty file in return (also attached).  The spice netlist is a simple 2-input CMOS NAND structure.

The netgen commands entered were as follows:

```
set filenum [readnet prm /usr/local/share/pdk/sky130A/libs.tech/irsim/sky130A_tt_nom_27.prm]
readnet spice untitled.spice $filenum
flatten untitled.spice
writenet sim untitled.spice
```

Are we doing something wrong?  Thank you!

[untitled.spice.txt](https://github.com/RTimothyEdwards/netgen/files/13505070/untitled.spice.txt)

[untitled.sim.txt](https://github.com/RTimothyEdwards/netgen/files/13505073/untitled.sim.txt)

RTimothyEdwards, 2023-12-01:
The issue here is that there is no subcircuit to be flattened.  The file "untitled.spice" is just a testbench and cell contents all combined into one file, and there is no ".subckt" definition.

The "flatten" command takes a subcircuit name as an argument.  "untitled.spice" is the cell name of the entire file---the top level---so flattening it is meaningless.  I guess in the best case I could have netgen detect that and complain about it.  Instead, in its attempt to flatten the top level, it is erasing it completely.

There are two simple solutions here.  The easiest one is to just remove the `flatten untitled.spice` line from the commands.  If you do that, you will also need to comment out all the ".include" lines in the `untitled.spice` input file because the parser is not prepared to read in the entire PDK and all of its SPICE model files.

The best practice, though, is to keep the subcircuit being designed and the testbench surrounding it separate.  Then you can create from xschem a netlist for only the subcircuit, not containing the testbench components (or the ".include" lines).  The subcircuit will be inside a ".subckt ... .ends" block (if the correct output option is selected in xschem), and then the command to flatten will be `flatten untitled` followed by `writenet sim untitled`.

The attached zip file shows what I mean, with two SPICE files, one being the testbench and the other being the subcircuit.
[untitled.zip](https://github.com/RTimothyEdwards/netgen/files/13519415/untitled.zip)

mhajimorad, 2023-12-02:
Thank you @RTimothyEdwards !  We will try your suggestions.  Thank you again!