RTimothyEdwards/netgen #86: 'M' Property Error on Single PNP

d-m-bailey, 2024-01-04:
netgen `1.5.264`

A single pnp transistor is extracted without an `M` parameter. Matching to a single pnp transistor in the source spice with an `m` parameter causes a mismatch.
Here’s the reduced layout:
```
X0 GND GND eg sky130_fd_pr__pnp_05v5_W0p68L0p68 M=8                               
X1 GND GND eu sky130_fd_pr__pnp_05v5_W0p68L0p68                                   
```
and here’s the schematic
```
XQ2 vss vss net2 sky130_fd_pr__pnp_05v5_W0p68L0p68 NE=1 m=8
XQ1 vss vss vn sky130_fd_pr__pnp_05v5_W0p68L0p68 NE=1 m=1
```
and here’s the error message
```
Netlists match uniquely with property errors.
Circuit 2 bandgap instance sky130_fd_pr__pnp_05v5_W0p68L0p68:Q1 property "m" has no match in circuit 1.
```

Currently, the `NE` parameter in the source is ignored in the test setup file. 
If the `NE` parameter is removed from the source, no warning message is generated.

To duplicate, export `PDK_ROOT` and `PDK` for sky130, then
```
1. tar xzf test-netgen.tgz
2. cd test-netgen
3. netgen -batch source lvs.script
```
[test-netgen.tgz](https://github.com/RTimothyEdwards/netgen/files/13824817/test-netgen.tgz)

You should be able to open the schematic `user_analog_project_wrapper.sch` in the `xschem` directory. 
Note that this is not the original schematic data but has been modified to (almost) match the layout.

RTimothyEdwards, 2024-01-04:
@d-m-bailey :  Thanks for the bug report and reproducible result.  I discovered the reason and why I could not duplicate the issue in a simple test.  Occasionally the "M" parameter gets recorded as type double (I forget the reason why), which was happening in this example, and that specific case was not being tested.  It is now fixed in netgen version 1.5.265.

d-m-bailey, 2024-01-04:
Excellent!

There was another property error in the test data, but that is due to a mismatch in the dummy mosfet counts and is not an issue.