RTimothyEdwards/netgen #88: netcmp.c / MatchPins

ymei, 2024-05-09:
I notice that the definition of `MatchPins()` in `netcmp.c` is guarded under `#ifdef TCL_NETGEN`, but its invocation by `spice.c` and `verilog.c` are not.  This is causing linking issues on my macOS (symbol not found) if `TCL_NETGEN` is not defined.

RTimothyEdwards, 2024-05-09:
Thank you for pointing this out.  The `#ifdef TCL_NETGEN` statement got moved to the wrong position and this error went unnoticed.

However, if you are not compiling with Tcl support, then you should be, since the non-Tcl version is missing a large amount of functionality.

If you want to ensure that you are compiling the right version under Mac OS, see the instructions at:

[http://opencircuitdesign.com/analog_flow/index.html](http://opencircuitdesign.com/analog_flow/index.html)

and specifically

[http://opencircuitdesign.com/analog_flow/appendix_3.html](http://opencircuitdesign.com/analog_flow/appendix_3.html)

ymei, 2024-05-09:
> Thank you for pointing this out. The `#ifdef TCL_NETGEN` statement got moved to the wrong position and this error went unnoticed.
>
> However, if you are not compiling with Tcl support, then you should [...]

Thanks.  Indeed, after figuring out `TCL_NETGEN,` I realized that I was missing tcl/tk related arguments for `configure`.