RTimothyEdwards/netgen #95: Enforce Port Order in LVS

sifferman, 2024-12-03:
Hello.

I noticed that LVS will pass for the following example:

```bash
netgen -batch \
 lvs "dff_model1.spice dff" "dff_model2.spice dff" ${PDK_ROOT}/sky130A/libs.tech/netgen/setup.tcl lvs.report -blackbox
```
```spice
* dff_model1.spice
.subckt dff CLK D Q
...
.ends
```
```spice
* dff_model1.spice
.subckt dff D CLK Q
...
.ends
```

Ideally, I would like it to fail because the port order is different. Is there a flag or parameter that would allow for this? I tried looking through the sky130_setup.tcl, but couldn't find anything.

<https://github.com/RTimothyEdwards/open_pdks/blob/master/sky130/netgen/sky130_setup.tcl>

Thanks!

d-m-bailey, 2024-12-03:
@sifferman normally port order is not checked in LVS. The circuits are identical regardless of the top level port order.

Where this can cause problems is if the extracted netlist is substituted verbatim for the source netlist in, say, a simulation test bench. Since spice is port order sensitive, this can cause problems if the ports are in a different order.

I believe that with magic, if you do a `readspice <spice_netlist>` as described [here](https://open-source-silicon.slack.com/archives/C01DH7SPX41/p1716062424821169?thread_ts=1715971254.413639&cid=C01DH7SPX41) the extracted ports will be in the same order as the `spice_netlist`. Be aware that the script shown is for a parasitic extraction which requires a flattened layout.

sifferman, 2024-12-04:
Thanks for the response. That is exactly the issue I am encountering: using cells in a testbench. However, I am not generating the spice file from magic. I am generating it from xschem.

This is not a problem, I just wrote a small check to run before LVS: <https://github.com/sifferman/sky130_schematics/blob/6f8764e4/Makefile#L33-L41>

Thanks!