RTimothyEdwards/netgen #96: Segfault when running an ihp LVS without the proper setup.tcl file in place

smunaut, 2025-03-08:
So while I was doing some work on iHP at some point I screwed up the setup.tcl and it ended up not referencing the proper PDK setup and it was causing a segfault.

I've attached the test case : 

[netgen.tar.gz](https://github.com/user-attachments/files/19142495/netgen.tar.gz)

`netgen -batch source lvs_script.lvs` to trigger the issue.

This crashes on my `nix-eda` setup that's using `Netgen 1.5.291 compiled on Thu Jan 16 11:42:32 UTC 2025` 

```
...
Circuit contains 11038 nets, and 18 disconnected pins.

Circuit 1 contains 113 devices, Circuit 2 contains 113 devices.
Circuit 1 contains 11038 nets,    Circuit 2 contains 11038 nets.

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff785b097 in PropertySortAndCombine () from /nix/store/8f8bnp1b5s5iqqspyhkn9hlxflwrqaaz-netgen-1.5.291/lib/netgen/tcl/tclnetgen.so
(gdb) bt
#0  0x00007ffff785b097 in PropertySortAndCombine () from /nix/store/8f8bnp1b5s5iqqspyhkn9hlxflwrqaaz-netgen-1.5.291/lib/netgen/tcl/tclnetgen.so
#1  0x00007ffff785d4db in PropertyMatch () from /nix/store/8f8bnp1b5s5iqqspyhkn9hlxflwrqaaz-netgen-1.5.291/lib/netgen/tcl/tclnetgen.so
#2  0x00007ffff785e00a in VerifyMatching () from /nix/store/8f8bnp1b5s5iqqspyhkn9hlxflwrqaaz-netgen-1.5.291/lib/netgen/tcl/tclnetgen.so
#3  0x00007ffff7885233 in _netcmp_verify () from /nix/store/8f8bnp1b5s5iqqspyhkn9hlxflwrqaaz-netgen-1.5.291/lib/netgen/tcl/tclnetgen.so
#4  0x00007ffff7885093 in _netcmp_run () from /nix/store/8f8bnp1b5s5iqqspyhkn9hlxflwrqaaz-netgen-1.5.291/lib/netgen/tcl/tclnetgen.so
#5  0x00007ffff7e2b218 in TclNRRunCallbacks () from /nix/store/ikq6xd0kxw4c3r812zaqvxjv2y8q40jc-tcl-8.6.15/lib/libtcl8.6.so
#6  0x00007ffff7e2d29d in TclEvalEx () from /nix/store/ikq6xd0kxw4c3r812zaqvxjv2y8q40jc-tcl-8.6.15/lib/libtcl8.6.so
#7  0x00007ffff7ee9819 in Tcl_FSEvalFileEx () from /nix/store/ikq6xd0kxw4c3r812zaqvxjv2y8q40jc-tcl-8.6.15/lib/libtcl8.6.so
#8  0x00007ffff7ee9aa7 in Tcl_EvalFile () from /nix/store/ikq6xd0kxw4c3r812zaqvxjv2y8q40jc-tcl-8.6.15/lib/libtcl8.6.so
#9  0x00007ffff7ef13ee in Tcl_SourceRCFile () from /nix/store/ikq6xd0kxw4c3r812zaqvxjv2y8q40jc-tcl-8.6.15/lib/libtcl8.6.so
#10 0x00007ffff7ef19c5 in Tcl_MainEx () from /nix/store/ikq6xd0kxw4c3r812zaqvxjv2y8q40jc-tcl-8.6.15/lib/libtcl8.6.so
#11 0x0000555555555201 in main ()
(gdb) 
```

On my workstation that's using a freshly compliled `Netgen 1.5.292 compiled on Sat Mar  8 08:53:20 AM CET 2025`, it doesn't crash but `valgrind` does show access to un-unitialized memory.

```

Circuit 1 contains 113 devices, Circuit 2 contains 113 devices.
Circuit 1 contains 11038 nets,    Circuit 2 contains 11038 nets.

==21245== Use of uninitialised value of size 8
==21245==    at 0x54D7B0F: PropertySortAndCombine (netcmp.c:4652)
==21245==    by 0x54DC6EC: PropertyMatch (netcmp.c:6180)
==21245==    by 0x54DD175: PropertyCheck (netcmp.c:6412)
==21245==    by 0x54DD422: VerifyMatching (netcmp.c:6483)
==21245==    by 0x550FB9A: _netcmp_verify (tclnetgen.c:2636)
==21245==    by 0x550F61C: _netcmp_run (tclnetgen.c:2488)
==21245==    by 0x48B142F: TclNRRunCallbacks (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x48B3448: TclEvalEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x496FFBA: Tcl_FSEvalFileEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x497023A: Tcl_EvalFile (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x4977A7F: Tcl_SourceRCFile (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x4977FEC: Tcl_MainEx (in /usr/lib64/libtcl8.6.so)
==21245== 
==21245== Use of uninitialised value of size 8
==21245==    at 0x54D7B1E: PropertySortAndCombine (netcmp.c:4653)
==21245==    by 0x54DC6EC: PropertyMatch (netcmp.c:6180)
==21245==    by 0x54DD175: PropertyCheck (netcmp.c:6412)
==21245==    by 0x54DD422: VerifyMatching (netcmp.c:6483)
==21245==    by 0x550FB9A: _netcmp_verify (tclnetgen.c:2636)
==21245==    by 0x550F61C: _netcmp_run (tclnetgen.c:2488)
==21245==    by 0x48B142F: TclNRRunCallbacks (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x48B3448: TclEvalEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x496FFBA: Tcl_FSEvalFileEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x497023A: Tcl_EvalFile (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x4977A7F: Tcl_SourceRCFile (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x4977FEC: Tcl_MainEx (in /usr/lib64/libtcl8.6.so)
==21245== 
==21245== Use of uninitialised value of size 8
==21245==    at 0x54EAA18: remove_group_tags (netgen.c:3141)
==21245==    by 0x54D7B4C: PropertySortAndCombine (netcmp.c:4661)
==21245==    by 0x54DC6EC: PropertyMatch (netcmp.c:6180)
==21245==    by 0x54DD175: PropertyCheck (netcmp.c:6412)
==21245==    by 0x54DD422: VerifyMatching (netcmp.c:6483)
==21245==    by 0x550FB9A: _netcmp_verify (tclnetgen.c:2636)
==21245==    by 0x550F61C: _netcmp_run (tclnetgen.c:2488)
==21245==    by 0x48B142F: TclNRRunCallbacks (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x48B3448: TclEvalEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x496FFBA: Tcl_FSEvalFileEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x497023A: Tcl_EvalFile (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x4977A7F: Tcl_SourceRCFile (in /usr/lib64/libtcl8.6.so)
==21245== 
==21245== Use of uninitialised value of size 8
==21245==    at 0x54EAA18: remove_group_tags (netgen.c:3141)
==21245==    by 0x54D7B60: PropertySortAndCombine (netcmp.c:4662)
==21245==    by 0x54DC6EC: PropertyMatch (netcmp.c:6180)
==21245==    by 0x54DD175: PropertyCheck (netcmp.c:6412)
==21245==    by 0x54DD422: VerifyMatching (netcmp.c:6483)
==21245==    by 0x550FB9A: _netcmp_verify (tclnetgen.c:2636)
==21245==    by 0x550F61C: _netcmp_run (tclnetgen.c:2488)
==21245==    by 0x48B142F: TclNRRunCallbacks (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x48B3448: TclEvalEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x496FFBA: Tcl_FSEvalFileEx (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x497023A: Tcl_EvalFile (in /usr/lib64/libtcl8.6.so)
==21245==    by 0x4977A7F: Tcl_SourceRCFile (in /usr/lib64/libtcl8.6.so)
==21245== 

Final result: 
Circuits match uniquely.
```

RTimothyEdwards, 2025-03-08:
I can duplicate your valgrind result, but I don't have the nix version and would have to figure out how to debug it (not sure if it is compiled with debugging symbols).  But valgrind is sometimes confused, especially about uninitialized variables, and this looks like a false positive to me.  Where valgrind flags the issues, everything looks properly allocated and set.

You can run gdb directly on netgen using (assuming the standard installation, not nix) `gdb /usr/local/lib/netgen/tcl/netgenexec` and then type the `source lvs_script.lvs` at the Tcl prompt.  I don't know if that's going to reveal anything more, though.

smunaut, 2025-03-08:
I did run gdb in nix as described above. The backtrace is posted above, right before the valgrind output.

The gdb backtrace is from the nix setup and shows crash in `PropertySortAndCombine` which is exactly where valgrind shows uninitialized reads.

RTimothyEdwards, 2025-03-08:
Yes, but it doesn't show any line numbers, which means that the nix version was compiled without debugging symbols, which makes it very hard to debug.  It's possible that the error gets surfaced by running compiler optimizations.

smunaut, 2025-03-08:
Ah yes. It crashes at ` ob2 = pre2->next;` because `pre2` is NULL.

RTimothyEdwards, 2025-03-08:
Okay, that's useful information.  But how can `pre2` be NULL?  If it was NULL, then `PropertyMatch()` should have caught it and returned immediately.  I'm missing something.

smunaut, 2025-03-09:
BTW, if I initialize `obn2` in `PropertyMatch` to NULL, then it crashes reliably even on my workstation with self-built netgen.

```diff
diff --git a/base/netcmp.c b/base/netcmp.c
index 23fe3ee..da39211 100644
--- a/base/netcmp.c
+++ b/base/netcmp.c
@@ -6010,7 +6010,7 @@ PropertyMatch(struct Element *E1, struct Element *E2,
                int do_print, int do_list, int *retval)
 {
    struct nlist *tc1, *tc2;
-   struct objlist *ob1, *ob2, *tp1, *tp2, *obn1, *obn2, *tpc;
+   struct objlist *ob1, *ob2, *tp1, *tp2, *obn1, *obn2 = NULL, *tpc;
    struct property *kl1, *kl2;
    struct valuelist *vl1, *vl2;
    struct NodeList *nl1, *nl2;
```

It doesn't get initialized by this loop : 

```
for (tp2 = ob2->next; (tp2 != NULL) && tp2->type > FIRSTPIN; tp2 = tp2->next)
      obn2 = tp2;
```

because the first `tp2->type == PROPERTY` ( which is value `-4` and thus fails the `> FIRSTPIN` test).

RTimothyEdwards, 2025-03-09:
@smunaut :  Thanks, that helps a lot.

RTimothyEdwards, 2025-03-09:
@smunaut :  This is a corner case which was missed;  the "pad" cell has only one pin, and it happens to have properties because two pads were combined in parallel, giving them an "M" property in the SPICE netlist.  The PropertyMatch() routine looks for the position in the cell contents (linked list) before the property, but the loop starts after the first pin, so if the first pin _is_ the position before the property, then it gets missed.  The solution is to set `obn1 = ob1->next;` and `obn2 = ob2->next;` before the loop starts at netcmp.c line 6060.  That both solves the uninizialized value condition and also fixes the error.

RTimothyEdwards, 2025-03-09:
The error has been corrected in netgen version 1.5.293.  I just pushed the fix to opencircuitdesign.com, and it will be mirrored to github by tomorrow.

smunaut, 2025-03-09:
Great thanks for the quick fix !

closed 2025-03-26 by RTimothyEdwards