RTimothyEdwards/netgen #97: Missing ports in netgen output.

d-m-bailey, 2025-05-15:
@RTimothyEdwards Any idea why backslashes cause ports to be dropped from the lvs comparison report?

Here's the netlist snippet
```
XRB1 \net1 VBB1 rhigh w=1.9u l=6u ps=0 b=0 m=1
XRB2 \net3 VBB1 rhigh w=1.9u l=5u ps=0 b=0 m=1
XRB3 \net5 VBB1 rhigh w=1.9u l=6u ps=0 b=0 m=1
XRB4 \net7 VBB2 rhigh w=2u l=6u ps=0 b=0 m=1
```
and here's the lvs log.
```
Subcircuit pins:
Circuit 1: rhigh                                                                  |Circuit 2: rhigh                                                                  
----------------------------------------------------------------------------------|----------------------------------------------------------------------------------
1                                                                                 |1                                                                                 
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Cell pin lists are equivalent.
Matching pins of cap_cmim in circuits 1 and 2
```

Normally, the `rhigh` devices would have two ports. Removing the backslashes from the netlist give the expected 2 port result.

If you don't know the reason off the top of your head, I will create a test case.

RTimothyEdwards, 2025-05-15:
It's probably due to netgen's parsing of nets for verilog, or something like that.  I would call it an error, though, since it seems easy enough to check and fix, so please create a test case.

d-m-bailey, 2025-05-15:
@RTimothyEdwards This reproduces the result for me using the `ihp_sg13g2` pdk.

```
#! /usr/bin/env bash

cat > netlist1.spice <<EOF
.SUBCKT LNA GND VBB1 VBB2 VCC RF_INPUT RF_OUTPUT 

XQ1 net2 net1 GND GND npn13g2 AE=0.063p PE=1.94u AB=25.605p PB=23.02u
+ AC=25.589984p PC=23.01u NE=4 m=4
XQ2 net4 net3 GND GND npn13g2 AE=0.063p PE=1.94u AB=25.605p PB=23.02u
+ AC=25.589984p PC=23.01u NE=4 m=4
XQ3 net6 net5 GND GND npn13g2 AE=0.063p PE=1.94u AB=12.988p PB=15.62u
+ AC=12.976684p PC=15.61u NE=2 m=2
XQ4 net8 net7 GND GND npn13g2 AE=0.063p PE=1.94u AB=12.988p PB=15.62u
+ AC=12.976684p PC=15.61u NE=2 m=2

XRC1 net2 VCC rsil w=7.5u l=5u ps=0 b=0 m=1
XRC2 net4 VCC rsil w=7.5u l=5u ps=0 b=0 m=1
XRC3 net6 VCC rsil w=7.5u l=5u ps=0 b=0 m=1
XRC4 net8 VCC rsil w=7.5u l=5.5u ps=0 b=0 m=1

XRB1 net1 VBB1 rhigh w=1.9u l=6u ps=0 b=0 m=1
XRB2 net3 VBB1 rhigh w=1.9u l=5u ps=0 b=0 m=1
XRB3 net5 VBB1 rhigh w=1.9u l=6u ps=0 b=0 m=1
XRB4 net7 VBB2 rhigh w=2u l=6u ps=0 b=0 m=1

XCC1 VCC GND cap_cmim w=20u l=25u A=500p P=90u m=8
XCB1 VBB1 GND cap_cmim w=20u l=25u A=500p P=90u m=6
XCB2 VBB2 GND cap_cmim w=20u l=25u A=500p P=90u m=4

.ENDS
EOF

cat > netlist2.spice <<EOF
.SUBCKT LNA GND VBB1 VBB2 VCC RF_INPUT RF_OUTPUT 

XQ1 \net2 \net1 GND GND npn13g2 AE=0.063p PE=1.94u AB=25.605p PB=23.02u
+ AC=25.589984p PC=23.01u NE=4 m=4
XQ2 \net4 \net3 GND GND npn13g2 AE=0.063p PE=1.94u AB=25.605p PB=23.02u
+ AC=25.589984p PC=23.01u NE=4 m=4
XQ3 \net6 \net5 GND GND npn13g2 AE=0.063p PE=1.94u AB=12.988p PB=15.62u
+ AC=12.976684p PC=15.61u NE=2 m=2
XQ4 \net8 \net7 GND GND npn13g2 AE=0.063p PE=1.94u AB=12.988p PB=15.62u
+ AC=12.976684p PC=15.61u NE=2 m=2

XRC1 \net2 VCC rsil w=7.5u l=5u ps=0 b=0 m=1
XRC2 \net4 VCC rsil w=7.5u l=5u ps=0 b=0 m=1
XRC3 \net6 VCC rsil w=7.5u l=5u ps=0 b=0 m=1
XRC4 \net8 VCC rsil w=7.5u l=5.5u ps=0 b=0 m=1

XRB1 \net1 VBB1 rhigh w=1.9u l=6u ps=0 b=0 m=1
XRB2 \net3 VBB1 rhigh w=1.9u l=5u ps=0 b=0 m=1
XRB3 \net5 VBB1 rhigh w=1.9u l=6u ps=0 b=0 m=1
XRB4 \net7 VBB2 rhigh w=2u l=6u ps=0 b=0 m=1

XCC1 VCC GND cap_cmim w=20u l=25u A=500p P=90u m=8
XCB1 VBB1 GND cap_cmim w=20u l=25u A=500p P=90u m=6
XCB2 VBB2 GND cap_cmim w=20u l=25u A=500p P=90u m=4

.ENDS
EOF

netgen -batch lvs "netlist1.spice LNA" "netlist2.spice LNA" $PDK_ROOT/$PDK/libs.tech/netgen/${PDK}_setup.tcl
```

RTimothyEdwards, 2025-05-18:
```
(gdb) print nexttok
$2 = 0x5584ad18fec0 "\\net2 \\net1"
```
The tokenizer pulls two net names at a time into a single token if there's a backslash.  Still investigating the mechanism.

RTimothyEdwards, 2025-05-18:
@d-m-bailey :  Fixed (pushed to opencircuitdesign.com, will mirror to github overnight, in about seven hours).  It was as I expected:  I had used the same base tokenizer for reading both SPICE and verilog, and the code in the parser dealing with verilog backslash notation was causing issues in this case.  I just added a flag argument to the parser indicating how it should handle backslashes.

RTimothyEdwards, 2025-05-18:
FYI, the fixed version is 1.5.295.

d-m-bailey, 2025-05-18:
@RTimothyEdwards Excellent!

closed 2025-05-18 by d-m-bailey